happycube/ld-decode #190: PAL low-MTF areas have herringbone effects...

happycube, 2019-05-05:
This is observed on good disks such as Jason.

happycube, 2019-05-05:
This is not a regression BTW, it is observed with rev4.  Still a serious bug.

happycube, 2019-06-06:
Oops, I didn't know this was open when I made #230.

simoninns, 2019-06-16:
With the newly repaired and calibrated LD-V4300D the sine-wave interference patterns are reduced on the Domesday discs (testing with set 1) - but still highly visible on National side 2 (which is CLV).  Here's frame 185 post comb-filter:

[image]

simoninns, 2019-06-16:
By way of experimentation I popped the Domesday AIV discs into a real Domesday system and took a look a the output; the wavy pattern is very obvious everywhere - so this is very likely an artefact of PAL discs common to all players which filter it well (like the LD-V4300D) or poorly (like the VP415).  Still not sure what actually causes it in the first place though.

happycube, 2019-12-21:
I think we figured out that some of this was calibration related, and the rest just the properties of the disk...

simoninns, 2019-12-22:
I'm still not convinced :)  But you can close the issue if you like.

closed 2020-07-30 by simoninns