happycube/ld-decode #191: Regression:  High # of spurious dropouts seen on some samples.

happycube, 2019-05-05:


Gamnn, 2019-05-05:
This seems to be a display bug in ld-analyse, not properly flushing the dropouts between TBCs. Peeking at a 9777 frame rotter TBC first, then opening a second TBC 58742 I get something like this on the second:

[image]

Quitting, then opening the second TBC on a fresh instance gives this:
[image]

simoninns, 2019-05-05:
Created an issue for the ld-analyse problem (as this is unrelated to the bug referenced by this issue) - issue #193

simoninns, 2019-05-05:
Domesday National disc set 2 side 1 frame 1500-1600 show the spurious drop-out detection issue well (if you need a sample to look at @happycube )

happycube, 2019-05-05:
Still investigating why, but the high frequencies on PAL demods before deemp are very high.  Last commit is a partial fix.

simoninns, 2019-12-22:
Roger Rabbit PAL CAV bonus disc seems to be another.  Frames 80-90 to around 300.

atsampson, 2020-01-01:
Another one: THX-1138 (PAL, TVL 002) side 2 frames 450-500, on brightly-coloured lettering. I've put a clip in sftp:adamsampson/Faults.

On the other hand, Pulp Fiction (NTSC, CC1423L) was suffering from this on neon signs (e.g. around side 1 frame 36560) back in September, but it's now decoding fine. I've put a clip of that there too.

simoninns, 2020-11-29:
This issue is really old for a bug... is it still a bug or has it been fixed? @happycube

happycube, 2020-11-29:
It should be better, but I'm not sure if it's *fixed*.

simoninns, 2020-11-30:
I suggest we close this then and, if someone has a working example of what's still not fixed, open a new issue with the recreation steps included.

closed 2020-11-30 by simoninns