happycube/ld-decode #199: Regression: field order output incorrect for Domesday National A

simoninns, 2019-05-09:
Performed a full disc decode of Domesday National A.  ld-analyse shows that the field order is not consistent:

file:
SFTP/LDV4300D_2 Captures/AIV/Domesday_DiscSet2/Domesday National_CAV_PAL_side1_Disc set 2_2018-10-02_18-14-36.lds

Switches:
--pal --EFM --start 357

Result (from ld-analyse):
First field flag order incorrect at frame 53894

simoninns, 2019-05-09:
Here are the git headers for the commit used for testing:

sdi@seholddecode01:~/github/ld-decode/.git$ more FETCH_HEAD 
7fca8174e5e08be66ac3598bc4afc9edffa9c5e5		branch 'rev5' of https:/
/github.com/happycube/ld-decode
sdi@seholddecode01:~/github/ld-decode/.git$ more ORIG_HEAD 
ba9bd3da2b7b1a0ac7da409f75f441efc8ebb846

simoninns, 2019-05-09:
Here is the debug from ld-analyse:

Debug: LdDecodeMetaData::read(): Found 107788 fields in the JSON document
Debug: LdDecodeMetaData::read(): Initial field has isFirstField = true
Debug: LdDecodeMetaData::read(): Field # 96980 has isFirstField out of sequence - TBC input file is broken
Debug: LdDecodeMetaData::read(): Field # 97037 has isFirstField out of sequence - TBC input file is broken
Debug: LdDecodeMetaData::read(): Field # 97186 has isFirstField out of sequence - TBC input file is broken
Debug: LdDecodeMetaData::read(): Field # 98445 has isFirstField out of sequence - TBC input file is broken
Debug: LdDecodeMetaData::read(): Field # 98622 has isFirstField out of sequence - TBC input file is broken
Debug: LdDecodeMetaData::read(): Field # 99121 has isFirstField out of sequence - TBC input file is broken
Debug: LdDecodeMetaData::read(): Field # 99798 has isFirstField out of sequence - TBC input file is broken
Debug: LdDecodeMetaData::read(): TBC file has 53898 first fields and 53890 second fields with 7 sequence errors
Debug: SourceVideo::open(): Called with field length = 355255
Debug: SourceVideo::open(): Successful - 107788 fields available

happycube, 2019-12-21:
Is this still an issue?

simoninns, 2019-12-21:
Don't think so :)

closed 2019-12-21 by simoninns