happycube/ld-decode #202: ld-decode producing invalid JSON parameters

simoninns, 2019-05-09:
When decoding the North Pole disc (full side 1 decode), ld-decode placed the text "Infinity" as the syncLevelPSNR (which should be a real number value, not a string).

Can you check this @happycube and ensure that only values befitting the field type are output into the JSON (otherwise this breaks the tool's shared library).

JSON from the decode is:

`
        {
            "isFirstField": false,
            "syncConf": 75,
            "seqNo": 51216,
            "diskLoc": 51757.2,
            "medianBurstIRE": 10.724614833395151,
            "dropOuts": {
                "fieldLine": [
                    137,
                    173,
                    224,
                    253
                ],
                "startx": [
                    286,
                    658,
                    526,
                    978
                ],
                "endx": [
                    343,
                    704,
                    559,
                    1001
                ]
            },
            "decodeFaults": 0,
            "vitsMetrics": {
                "greyPSNR": 39.756,
                "greyIRE": -4.252,
                "palVITSBurst50Level": 1.001,
                "whiteSNR": 35.498,
                "whiteIRE": 95.052,
                "whiteRFLevel": 8854.506,
                "blackLineRFLevel": 9212.575,
                "blackLinePreTBCIRE": -4.255,
                "blackLinePostTBCIRE": -3.966,
                "blackLinePSNR": 38.765,
                "syncLevelPSNR": Infinity,
                "syncRFLevel": 9151.323,
                "syncToBlackRFRatio": 0.993,
                "syncToWhiteRFRatio": 1.034,
                "blackToWhiteRFRatio": 1.04
            },
            "frameNumber": 25607,
            "audioSamples": 962
        },

`

oyvindln, 2019-05-09:
I've ran into this when working on VHS too, some of the metrics calculations can end up trying to divide by zero or take the logarithm of a negative number, so they should be verified before writing to JSON.

happycube, 2019-05-09:
Should be fixed now

commit b6a57706 (2019-05-09): fix #202, some work on #199
closed 2019-05-19 by happycube