happycube/ld-decode #205: Multiple dropouts on one line can cause DOD to mark good spots

Gamnn, 2019-05-10:
Seems to be the case more for longer dropouts, but having more than one on a line causes the DOD to mark a lot of the good portion between them.

[greedydod.lds.zip](https://github.com/happycube/ld-decode/files/3164419/greedydod.lds.zip)

[image]
[image]

happycube, 2019-05-21:
(note, may slip to rev6)

happycube, 2019-06-01:
Put a cap on it's greedyness (which is quite useful in areas with long dropouts... cutting it completely causes visible false negatives).  I'll mark this done until/unless there's a nasty enough sample to re-evaluate my life^Wapproach.

closed 2019-06-01 by happycube