happycube/ld-decode #206: Solid red areas on NTSC can have herringbone effect

Gamnn, 2019-05-13:
Possibly related to #190, merge if so.

Tested on rev4, issue exists there as well.

[ntscherring.lds.zip](https://github.com/happycube/ld-decode/files/3172534/ntscherring.lds.zip)

Frame 3 ld-decode output:
[image]

s-video capture:
[image]

happycube, 2019-05-31:
It's regressed and now vrooping

Gamnn, 2019-05-31:
vroop regression caused by ee8a5c1299a1882cf4cb0a36dc9d9a5f7452b96d.

happycube, 2019-05-31:
oopsie.  backed out unintentional commit.

happycube, 2019-06-06:
I'm playing around with RF decoding parameters and they help a lot.  So a new option for less agressive/sharp settings is a good idea...
---
ldd.rf.DecoderParams['video_bpf'] = (3500000, 12500000)
ldd.rf.DecoderParams['video_bpf_order'] = 4

ldd.rf.DecoderParams['video_lpf_freq'] = (4200000)
ldd.rf.DecoderParams['video_lpf_order'] = 6
---

happycube, 2019-06-08:
Added a new --video-bpf-high parameter (in mhz) to ld-decode.py just now.  "--video_bpf_high 12.2" squelches it on this disk.

simoninns, 2019-06-08:
It would be good to generate a wiki page for all these additional options (and explain briefly what they are for as well as the range of usable settings)

happycube, 2019-06-09:
Added new page "Fine tuning parameters" but there should be a troubleshooting page as well.

happycube, 2019-06-16:
Closing this, at least for now :)

Gamnn, 2019-12-30:
On some sources, adjusting the bpf doesn't clear up the wibble before considerably weakening the chroma and washing out colors.

Attaching some more samples across a variety of discs and players.

[akowibble-a-x9.zip](https://github.com/happycube/ld-decode/files/4009004/akowibble-a-x9.zip)
[akowibble-b-x9.zip](https://github.com/happycube/ld-decode/files/4009005/akowibble-b-x9.zip)
[akowibble-c-x9.zip](https://github.com/happycube/ld-decode/files/4009006/akowibble-c-x9.zip)
[akowibble-eb-dvl919.zip](https://github.com/happycube/ld-decode/files/4009007/akowibble-eb-dvl919.zip)
[akowibble-e-dvl919.zip](https://github.com/happycube/ld-decode/files/4009008/akowibble-e-dvl919.zip)
[akowibble-r-clds315.zip](https://github.com/happycube/ld-decode/files/4009009/akowibble-r-clds315.zip)

happycube, 2019-12-30:
Still experimenting - the below RFParams block helps with akowibble-a-x9.ldf and the majo x9 sample but doesn't eliminate the wibble.

In the case of majo there's a spurious signal at the burst frequency and a harmonic around the main wave at 4.8mhz.  ld-decode's defaults are sharp enough to pick this up.

```
RFParams_NTSC = {
    # The audio notch filters are important with DD v3.0+ boards
    'audio_notchwidth': 350000,
    'audio_notchorder': 2,

    'video_deemp': (120*.32, 320*.32),

    # This BPF is similar but not *quite* identical to what Pioneer did
    'video_bpf_low': 4000000, 
    'video_bpf_high': 13000000,
    'video_bpf_order': 4,

    # This can easily be pushed up to 4.5mhz or even a bit higher. 
    # A sharp 4.8-5.0 is probably the maximum before the audio carriers bleed into 0IRE.
    'video_lpf_freq': 4200000,   # in mhz
    'video_lpf_order': 6, # butterworth filter order

    # MTF filter
    'MTF_basemult': .4, # general ** level of the MTF filter for frame 0.
    'MTF_poledist': .9,
    'MTF_freq': 12.2, # in mhz

    # used to detect rot
    'video_hpf_freq': 10000000,
    'video_hpf_order': 4,

    'audio_filterwidth': 150000,
    'audio_filterorder': 800,
}

```

Gamnn, 2020-02-10:
Between --WibbleRemover and --lowband, and the manual high/low pass filters, I think this can be closed now. The wiki could use an explanation of the options and when/how to use them however.

simoninns, 2025-12-09:
Closed as fixed

commit 139f7c1c (2019-06-08): fix #206... by adding new parameters