happycube/ld-decode #211: Add 3D comb filter support to ld-comb-pal

simoninns, 2019-05-19:
The PAL comb filter is currently 2D only.  It would be good to add support for 3D comb filtering

atsampson, 2019-06-04:
Worth noting the previous links about the BBC Transform PAL decoder in #9, including Jim Easterbrook's GPLv3 implementation.

simoninns, 2019-12-22:
Updated issue title according to the rev6 naming

happycube, 2020-09-27:
Does the transform decoder count for this, or should there be a non-transform 3D decoder as well?

simoninns, 2020-09-27:
No idea.  My assumption was: issue open = issue not solved.  @atsampson - please close if issue closed or explain what issue is :)

atsampson, 2020-09-27:
I'd say transform3d meets this requirement. I guess we could also implement an adaptive 3D PAL decoder that works like ntsc3d, but I'm not sure there'd be much point... at present I think the best thing we could do to improve PAL quality further is to figure out a way of doing automatic video equalisation so we could use the 1D demodulator by default with transform3d (i.e. the VITS work).

closed 2020-09-27 by atsampson