happycube/ld-decode #217: ld-decode crash

Gamnn, 2019-05-19:
Threw one of my worst rotters at it, didn't get far without a crash:

```
$ python3 ~/github/ld-decode-rev5/ld-decode.py -nj rottest_1.lds rottest1
file frame 0 CAV frame 341
file frame 1 CAV frame 342
Traceback (most recent call last):
  File "~/github/ld-decode-rev5/ld-decode.py", line 123, in <module>
    write_json(ldd, outname)
  File "~/github/ld-decode-rev5/ld-decode.py", line 89, in write_json
    jsondict = ldd.build_json(ldd.curfield)
  File "~/github/ld-decode-rev5/lddecode_core.py", line 2338, in build_json
    vp['black16bIre'] = np.float(f.hz_to_output(f.rf.iretohz(self.blackIRE)))
  File "~/github/ld-decode-rev5/lddecode_core.py", line 1643, in hz_to_output
    return np.uint16(np.clip((reduced * self.out_scale) + 1024, 0, 65535) + 0.5)
AttributeError: 'FieldNTSC' obj

```
[rottest_1.lds.zip](https://github.com/happycube/ld-decode/files/3195605/rottest_1.lds.zip)

Gamnn, 2019-05-19:
Still crashes:

```
python3 ~/github/ld-decode/ld-decode.py -nj rottest_1.lds rottest1
file frame 0 CAV frame 341
file frame 1 CAV frame 342
WARNING:root:Severe VSYNC-area corruption detected.
ERROR - please paste the following into a bug report:
Traceback (most recent call last):
  File "~/github/ld-decode/ld-decode.py", line 102, in <module>
    f = ldd.readfield()
  File "~/github/ld-decode/lddecode_core.py", line 1954, in readfield
    f, offset = self.decodefield()
  File "~/github/ld-decode/lddecode_core.py", line 1932, in decodefield
    f = self.FieldClass(self.rf, self.indata, self.rawdecode, audio_offset = self.audio_offset, prevfield = self.curfield)
  File "~/github/ld-decode/lddecode_core.py", line 1672, in __init__
    super(FieldNTSC, self).__init__(*args, **kwargs)
  File "~/github/ld-decode/lddecode_core.py", line 1232, in __init__
    self.linelocs1, self.linebad, self.nextfieldoffset = self.compute_linelocs()
  File "~/github/ld-decode/lddecode_core.py", line 966, in compute_linelocs
    line0loc, self.isFirstField = self.getLine0(validpulses)
  File "~/github/ld-decode/lddecode_core.py", line 928, in getLine0
    return self.prevfield.linelocs[self.prevfield.outlinecount - 1] - self.prevfield.nextfieldoffset, not self.prevfield.isFirstField
AttributeError: 'FieldNTSC' object has no attribute 'linelocs'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "~/github/ld-decode/ld-decode.py", line 109, in <module>
    print("current sample: " + ldd.fdoffset, file=sys.stderr)
TypeError: can only concatenate str (not "numpy.int64") to str
Info: Processing complete
```

[rottest_1-10frame.lds.zip](https://github.com/happycube/ld-decode/files/3195677/rottest_1-10frame.lds.zip)

happycube, 2019-05-30:
Disk no longer crashes, just has a tiny bit of vroop.  It turns out for some reason the disk is 10+IRE out of spec frequency wise possibly because of it's rot... closing

commit 51574716 (2019-05-25): fix #217 and strengten framing code some more
closed 2019-05-30 by happycube