happycube/ld-decode #219: ld-process-vbi eats .json on failure

Gamnn, 2019-05-20:
Likely related to #201, ld-process-vbi can fail to write the .json on long .TBCs, overwriting the original with a 0-byte file. Suggested workaround is for any tools that replace an existing .json to write to .json.tmp and mv it to replace the original only on success. 

```
Info: Processing field 164400 16 = "8dc000" 17 = "87ffff" 18 = "881ddd"
Info: Processing field 164401 16 = "0" 17 = "80eeee" 18 = "80eeee"
Info: Processing field 164402 16 = "0" 17 = "80eeee" 18 = "80eeee"
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
```

Gamnn, 2019-05-20:
Cause for the crash is likely lack of RAM in the box I usually run the tools on (4GB). Running it on another machine with 16GB completes the same .tbc successfully, using up to around 2GB while it prepares to write. My wallet would be thankful if it wrote incrementally to reduce the memory load.

simoninns, 2019-05-21:
Ubuntu or another OS?

As a first step #201 should be fixed.  It much quicker to make the JSON output from ld-decode a sane amount than it is to re-code the JSON library to compensate for the big metadata files.

Gamnn, 2019-05-21:
On Ubuntu.

simoninns, 2019-06-06:
Modified ld-decode-tools shared library so (all tools) will:

a) write the new JSON to a system temporary file
b) rename the original file to filename+.tmp
c) move the temporary file to the original filename
d) delete filename+.tmp
e) delete the system temporary file

quite a 'belt and braces' approach to the issue I think :)

closed 2019-06-06 by simoninns