happycube/ld-decode #220: EFM decoding fail

Gamnn, 2019-05-24:
ld-decode or ld-process-efm fails on this .lds in an interesting spot- there doesn't seem to be any dropout and the SNR is good.  Doesn't appear to be a regression with the new framing code- broken on old commits too.

[issue220_efmbroke.lds](https://drive.google.com/open?id=1yLsHE71rdftuuV0uVHklpRCkAB6cLefk)

```
$ ld-process-efm -i broke.efm -a broke.digital.pcm -dx
Debug: EfmProcess::openInputFile(): 10-bit input file is "broke.efm" and is 881024 bytes in length
Debug: F2FramesToAudio::openOutputFile(): Opened "broke.digital.pcm" as audio output file
Debug: EfmToF3Frames::sm_state_findInitialSyncStage1(): Initial F3 sync found at buffer position 14
Debug: EfmToF3Frames::sm_state_findInitialSyncStage2(): Found first F3 frame with a length of 587 bits
Debug: EfmToF3Frames::sm_state_processFrame(): F3 Frame 1 frame length incorrect T = 587
Debug: EfmToF3Frames::sm_state_findSecondSync(): F3 Sync value low and off by one transition backwards
Debug: EfmToF3Frames::sm_state_processFrame(): F3 Frame 2 frame length incorrect T = 587
Debug: SectionToMeta::process(): Section Q mode 4 - LD Audio - Audio (Encoding running): Track = 1 - Subdivision = 1 - Time = "04:08.50" - Disc Time = "04:10.50"
Debug: SectionToMeta::process(): Section Q mode 4 - LD Audio - Audio (Encoding running): Track = 1 - Subdivision = 1 - Time = "04:08.51" - Disc Time = "04:10.51"
Debug: SectionToMeta::process(): Section Q mode 4 - LD Audio - Audio (Encoding running): Track = 1 - Subdivision = 1 - Time = "04:08.52" - Disc Time = "04:10.52"
Debug: SectionToMeta::process(): Section Q mode 4 - LD Audio - Audio (Encoding running): Track = 1 - Subdivision = 1 - Time = "04:08.53" - Disc Time = "04:10.53"
Debug: SectionToMeta::process(): Section Q mode 4 - LD Audio - Audio (Encoding running): Track = 1 - Subdivision = 1 - Time = "04:08.54" - Disc Time = "04:10.54"
Debug: SectionToMeta::process(): Section Q mode 4 - LD Audio - Audio (Encoding running): Track = 1 - Subdivision = 1 - Time = "04:08.55" - Disc Time = "04:10.55"
Info: Processed 9% of the input EFM
Debug: EfmToF3Frames::sm_state_findSecondSync(): F3 Sync value low and off by one transition backwards
Debug: EfmToF3Frames::sm_state_processFrame(): F3 Frame 722 frame length incorrect T = 579
Debug: EfmToF3Frames::sm_state_findSecondSync(): F3 frame length was incorrect ( 590 ), but error is less than T3, so nothing much to do about it
Debug: EfmToF3Frames::sm_state_processFrame(): F3 Frame 723 frame length incorrect T = 590
Debug: EfmToF3Frames::sm_state_findSecondSync(): F3 frame length was incorrect ( 592 ), moving end transition in attempt to correct
Debug: EfmToF3Frames::sm_state_processFrame(): F3 Frame 724 frame length incorrect T = 584
Debug: EfmToF3Frames::sm_state_findSecondSync(): F3 frame length was incorrect ( 590 ), but error is less than T3, so nothing much to do about it
Debug: EfmToF3Frames::sm_state_processFrame(): F3 Frame 725 frame length incorrect T = 590
Debug: EfmToF3Frames::sm_state_findSecondSync(): F3 frame length was incorrect ( 589 ), but error is less than T3, so nothing much to do about it
Debug: EfmToF3Frames::sm_state_processFrame(): F3 Frame 726 frame length incorrect T = 589
Debug: EfmToF3Frames::sm_state_findSecondSync(): F3 frame length was incorrect ( 591 ), moving end transition in attempt to correct
Debug: EfmToF3Frames::sm_state_processFrame(): F3 Frame 728 frame length incorrect T = 584
Debug: EfmToF3Frames::sm_state_findSecondSync(): F3 frame length was incorrect ( 589 ), but error is less than T3, so nothing much to do about it
Debug: EfmToF3Frames::sm_state_processFrame(): F3 Frame 730 frame length incorrect T = 589
Debug: EfmToF3Frames::sm_state_findSecondSync(): F3 frame length was incorrect ( 595 ), moving end transition in attempt to correct
Debug: EfmToF3Frames::sm_state_processFrame(): F3 Frame 731 frame length incorrect T = 586
Debug: EfmToF3Frames::sm_state_findSecondSync(): F3 frame length was incorrect ( 592 ), moving end transition in attempt to correct
Debug: EfmToF3Frames::sm_state_processFrame(): F3 Frame 733 frame length incorrect T = 587
Debug: EfmToF3Frames::sm_state_findSecondSync(): F3 frame length was incorrect ( 590 ), but error is less than T3, so nothing much to do about it

...

Info: EFM to F3 Frame converter:
Info:   Total number of F3 Frames = 7235
Info:   of which 5321 were 588 bits and 1914 were invalid lengths
Info:   Lost frame sync 0 times
Info: 
Info: C1 Level error correction:
Info:   Total number of C1s processed = 7234
Info:   of which 895 passed and 6339 failed
Info:   The C1 error correction recovered 241 corrupt C1s
Info:   The delay buffer was flushed 0 times
Info: C2 Level error correction:
Info:   Total number of C2s processed = 7126
Info:   of which 674 passed and 6452 failed
Info:   The C2 error correction recovered 59 corrupt C2s
Info:   The delay buffer was flushed 0 times
Info: C2 Deinterleave:
Info:   Total number of C2s processed = 7124
Info:   of which 627 were valid and 6497 were invalid
Info:   The delay buffer was flushed 0 times
Info: 
Info: F2 to F1 converter:
Info:   Total number of F1 frames = 0
Info:   Number of F1 frames with sync missing = 0
Info:   Lost F1 sync 0 times
Info: 
Info: F2 Frames to audio converter:
Info:   Total number of stereo audio samples = 7124
Info: 
Info: F3 to section converter:
Info:   Total number of sections = 11
Info:   Number of sections with SYNC0 or SYNC1 missing = 4
Info:   Lost section sync 1 times
Info: 
Info: Sections (audio/data) to metadata processing:
Info:   Total number of sections processed = 11 ( 1078 F3 frames )
Info:   Q Mode 0 sections = 0 (Data)
Info:   Q Mode 1 sections = 0 (CD Audio)
Info:   Q Mode 2 sections = 0 (Disc ID)
Info:   Q Mode 3 sections = 0 (Track ID)
Info:   Q Mode 4 sections = 7 (LD Audio)
Info:   Sections with failed Q CRC = 4
Info: 
Info:   Sections with qControl flagged as audio = 7
Info:   Sections with qControl flagged as data = 4
Info:   Total qControl sections = 11
Info: 
Debug: SectionToMeta::closeOutputFile(): Processing 11 metadata entries to JSON
Debug: SectionToMeta::closeOutputFile(): Writing JSON metadata file
```

In the full decode, it recovers after 7 minutes.
[image]

Gamnn, 2019-05-25:
Whatever's breaking it seems to be in fileframe 2 of issue220_efmbroke.lds. starting at -s 3 or higher produces valid output.

```
Info: C2 Deinterleave:
Info:   Total number of C2s processed = 6387
Info:   of which 6387 were valid and 0 were invalid
Info:   The delay buffer was flushed 0 times
```

Reminds me of #182, but without any discernible error on the TBC. Unlike 182, output directly from ld-ldstoefm seems to break in the same spot, so I think it may be a problem in ld-ldstoefm or ld-process-efm.

happycube, 2019-05-25:
Two issues: #222 and recovery from a bad sample.  I suspect that it should recover within seconds at worst in a case like this.  The question is what needs to be strengthened...

simoninns, 2019-07-14:
I think this has already been covered by the work around ld-decode-efm and the changes to the EFM filter stages.  Closing issue (feel free to open new issues with up-to-date tests against the new commits if required)

closed 2019-07-14 by simoninns