happycube/ld-decode #221: ld-analyse dropout analysis visual glitch

Gamnn, 2019-05-24:
When the .tbc has no dropouts, it causes a glitch in the dropout analysis chart window.

Ubuntu:
[image]

Mac:
[image]

Gamnn, 2019-06-02:
New oddity, 0 is on the graph twice:

[image]
[image]

simoninns, 2019-06-06:
Fixed both issues and also fixed 'double zero' issue in the SNR graph too.

closed 2019-06-06 by simoninns