happycube/ld-decode #225: ld-combine - add option to prefer sub-line merging

Gamnn, 2019-05-27:
If I understand correctly, ld-combine prefers to replace a full line in the case of a dropout, except in cases where both .tbcs have dropouts in different parts of the same line, in which case it replaces just the dropout section.

It would be preferable to only do the latter in other cases as well, such as when preparing .tbcs for median stacking, so an option to disable full-line replacement would be appreciated.

simoninns, 2019-05-27:
ld-combine already does both full and sub-line replacement.  It will only replace the whole line if the source doesn't have any drop-outs on the line.  If it's copying over lines with dropouts it's probably because ld-decode didn't mark DOs on the source (it's missing quite a lot of DOs at the moment).  

I can only really look into improving ld-combine once ld-decode has less missed DOs, so I'll leave this open until then.

Gamnn, 2019-05-27:
> ld-combine already does both full and sub-line replacement. It will only replace the whole line if the source doesn't have any drop-outs on the line.

Right, this request is an option to disable that behavior and replace only the dropout area instead of full line. If you have sources with differing SNR for example, you would not want to replace a mostly good line with one full of noise just for a single dropout, but just replace the part with the dropout. There's other use cases as well.

simoninns, 2019-05-27:
That issue should be solved by #122 which I generated to automate the issue you are describing

Gamnn, 2019-05-27:
Perhaps I should really be requesting that median stacking be implemented in the tool directly, as such changes would reduce its effectiveness run separately. I think it could be ported from imagemagick.
I may raise another issue once I've had more time to experiment.

Gamnn, 2019-06-05:
Related, it appears sub-line replacements are broken at the moment. The wrong offsets are replaced.
For testing, [I removed the full-line replacement code, forcing sub-line](https://github.com/Gamnn/ld-decode/commit/9203a8c231bb1d64fc813405b4f8e046b3684f6f?diff=unified)

https://slowpics.org/comparison/5615baf3-a023-4136-a664-bbfbd32109e7

closed 2019-06-27 by simoninns