happycube/ld-decode #226: Add .lds compression and decompression scripts to the repo

simoninns, 2019-06-01:
The .lds compression and decompression 'best-practice' commands should be added as scripts to the repo to make the process easier

simoninns, 2019-06-01:
Some notes from Matt_O on IRC:

.LDS to FLAC (with md5):
-----------------------------
pv firefox.lds | tee >(dddconv -u | ffmpeg -y -f s16le -ar 40k -ac 1 -i - -acodec flac -compression_level 11 firefox.raw.oga 2>/dev/null ) | md5sum > firefox.lds.md5.txt
 
===================================================================================
 
FLAC to .LDS (with md5):
---------------------
ffmpeg -i ralf.raw.oga -f s16le -c:a pcm_s16le - | dddconv -p | tee ralf.lds | md5sum
 
===================================================================================
 
To check md5 without decoding (saves time by not writing to disk):
-----------------------------------------------------------------
 
ffmpeg -i ralf.raw.oga -f s16le -c:a pcm_s16le - | dddconv -p | md5sum
 
===================================================================================
 
 
To split .LDS files to prepare to compress/backup on BD-R:
-------------------------------------------------------
pv big.lds | split -d -b 40GB - big.lds.
 
NOTES:
- Converting to FLAC should be done on each split file
- To ensure that data can be partially recovered if needed, split the source file on 5 byte boundaries.  Therefore, use 'GB' instead of 'G' for the 'split' util size argument.
- Compression is assumed to make file 0.55 times as small as source.
    Therefore a 40GB source file would compress down to 22GB.
    This leaves enough extra space on a 25GB BD-R for some .par2 recovery files and file system overhead.
    Tweak according to your circumstances (for example, you may need to use a smaller split size if you do not obtain as good of compression)
 
To re-assemble split files:
 
cat big.lds.* > big.lds
 
Use md5sum to verify that the source file can be recovered before deleting the source file :)
 
==============================================================================================
 
To encode viewable PAL video file:
 
ffmpeg -y -f s16le -ar 48k -ac 2 -i dl.pcm -f rawvideo -r 25 -pix_fmt rgb48 -s 928x576 -i dl.rgb -aspect 4:3 -vcodec ffv1 -acodec flac -pix_fmt yuv422p dl.mkv

simoninns, 2019-06-01:
Options should be 'compress', 'decompress', 'verify' and 'split'

Gamnn, 2019-06-01:
FLAC has md5 hashes built-in, and ffmpeg will report if it fails on decompression, so running md5sum is likely over-kill (though I still do it anyway).

Here's the batch compression script I use (adapted from someone elses')
`for f in *.lds ; do dddconv -u -i "$f" |  ffmpeg -f s16le -ar 40k -ac 1 -i - -compression_level 11 "${f%.*}.raw.oga" ; done && cfv -C -v -t md5`

This compresses each .lds in the directory, then generates a .md5 file for all of the files in the directory (requires [cfv](http://cfv.sourceforge.net/)).

Another note, you can convert a snip from FLAC to .lds as well by multiplying the desired time by 1000, using ffmpeg's git head (current stable cannot seek past 99 hours.)

Example, cut a 1 minute clip from 24 minutes in:
24 minutes * 1000 = 400 hours
1 minute * 1000 = 16 hours 40 minutes
`ffmpeg -ss 400:00:00 -i big.raw.oga -t 16:40:00 -f s16le -c:a pcm_s16le - | dddconv -p -o clip.lds`
Or
`ffmpeg -ss 400:00:00 -i big.raw.oga -to 416:40:00 -f s16le -c:a pcm_s16le - | dddconv -p -o clip.lds`

Gamnn, 2019-06-27:
#265 and #266 ready for review. Only tested on Mac so far.

Gamnn, 2019-06-27:
Did some testing on the built-in flac md5; it seems ffmpeg doesn't write it to .oga files (should be in the STREAMINFO). It does write them to .flac files.

On the decode side, ffmpeg fails to catch checksum failures, though flac catches them.

Overall, it's best practice to md5sum the original .lds and .raw.ogas. I'll write another script for the process of checksumming and verifying.

simoninns, 2019-07-02:
Fixed in current rev6

closed 2019-07-02 by simoninns