happycube/ld-decode #229: v5 halts on ld-decode.py, but v4 does not.

toynbeeidea, 2019-06-03:
I have an LDS file I capped a while ago, and I successfully decoded it with v4.  When I attempt to do so with v5, however, nothing happens:  Like, no error messages, no output, just a blinking cursor and a pinned CPU core.  Zero-length efm, pcm, and tbc files are created.  If I ctrl-C out of it, a small json is generated:

```
{
    "pcmAudioParameters": {
        "bits": 16,
        "isLittleEndian": true,
        "isSigned": true,
        "sampleRate": 48000
    },
    "videoParameters": {
        "numberOfSequentialFields": 0,
        "isSourcePal": false,
        "fsc": 3579545,
        "fieldWidth": 910,
        "sampleRate": 14318180,
        "black16bIre": 15360.0,
        "white16bIre": 51200.0,
        "fieldHeight": 263,
        "colourBurstStart": 74.0,
        "colourBurstEnd": 110.0,
        "activeVideoStart": 134.0,
        "activeVideoEnd": 887.0
    },
    "fields": []
}
```
But that's all I get.

simoninns, 2019-06-04:
At a first guess, I'd say the LDS file doesn't contain any data (cable disconnected during capture or something similar).  Use dddutil to convert a snippet of the file to 16-bit data and then look at it in audacity to confirm there is valid RF sample data.

toynbeeidea, 2019-06-04:
https://drive.google.com/open?id=1vmZiQowin4gF8OJNuCC6bNVzyo8LKfdn

While trimming  this out  and uploading it I left the python ld-decode command running (python3 ld-decode.py sanctuary.lds sanctuary --NTSCJ --ignoreleadout), and it eventually did begin to decode - but it took quite a long time.  I began this particular capture with a paused player becuase #177 was still an issue at the time, so I guess there's a lot of "junk frames" at the beginning.  Now that I think about it, I probably used -s to skip those in v4 due to crashing.  v5 doesn't crash, it just plows ahead until it find something it can work with - its just that it gives no indication that its doing so, and given how slow ld-decode currently is, well... I'm sure you can see where I kinda panicked.  

So I guess the issue isn't so much of a "halt" as it is a lack of feedback.

simoninns, 2019-06-05:
The best practice at the moment (or at least what I use) is to use --start 500 --length 3 to skip over any un-decodable spin-up data.  ld-decode should respond with the frame number/time code at file position 500; you can then subtract the difference from 500 to get the actual start point of the first frame (and then loose the --length option).

It would be better if ld-decode wasn't silent about skipping undecodable data though, so I will change the title of this issue and leave it open for now.

happycube, 2019-06-27:
It'll say something now...

closed 2019-06-27 by happycube