happycube/ld-decode #231: ld-dropout-correct segfault

toynbeeidea, 2019-06-07:
I decoded a double-sided disc using the --ignoreleadout flag.  That went well (as near as I can tell so far).  But when I ran ld-dropout-correct, not so much.
```
Info: Frame # 109997 [ 219994 / 219995 ] - 0 dropouts corrected
Info: Frame # 109998 [ 219996 / 219997 ] - 0 dropouts corrected
Info: Frame # 109999 [ 219998 / 219999 ] - 0 dropouts corrected
Critical: Attempting to get field number failed - no isFirstField in JSON before end of file
Critical: LdDecodeMetaData::getField(): Requested field number -1 out of bounds!
Critical: Attempting to get field number failed - no isFirstField in JSON before end of file
Critical: LdDecodeMetaData::getField(): Requested field number -1 out of bounds!
Critical: LdDecodeMetaData::getField(): Requested field number -1 out of bounds!
Critical: LdDecodeMetaData::getField(): Requested field number -1 out of bounds!
Warning: Requested field number -1 is out of range!
Segmentation fault (core dumped)
```

simoninns, 2019-06-07:
Right now capturing a double-sided disc in one go should be considered unsupported.  The tool chain is tested to the limits of the IEC specification (around 80K frames iirc).

Suggested work around would be to capture each side separately.  If you have a lot of discs to capture and need to save time, it's probably a good idea to look into the automated capture features in the DdD GUI (and use a compatible player).  Also, turning off side changing and using a capture duration limit is another less convenient method.

It's possible that issue #103 is the root-cause too; might be worth retesting once that is fixed.

toynbeeidea, 2019-06-07:
https://drive.google.com/open?id=1WOLhSBF_CLhiMlXtxCCDGET1wP6PUVZA - if you need/want it.

For now I'll just chop up the existing lds file and reprocess separately.

simoninns, 2019-07-13:
I've made some fixes to the ld-decode-shared library code which should either correct this issue (or, if it's really because a firstField seq. error from ld-decode) handle it better

closed 2019-07-13 by simoninns