happycube/ld-decode #237: Crash on seeking

toynbeeidea, 2019-06-11:
First time seeing one of these in v5, but...

```
>python3 ld-decode.py sanctuary.lds sanctuary-side2 --NTSCJ -s 59880

ERROR - please paste the following into a bug report:
Traceback (most recent call last):
  File "ld-decode.py", line 102, in <module>
    f = ldd.readfield()
  File "/home/erik/NASSync/ld-decode.5.21.19/lddecode_core.py", line 1958, in readfield
    f, offset = self.decodefield()
  File "/home/erik/NASSync/ld-decode.5.21.19/lddecode_core.py", line 1936, in decodefield
    f = self.FieldClass(self.rf, self.indata, self.rawdecode, audio_offset = self.audio_offset, prevfield = self.curfield)
  File "/home/erik/NASSync/ld-decode.5.21.19/lddecode_core.py", line 1676, in __init__
    super(FieldNTSC, self).__init__(*args, **kwargs)
  File "/home/erik/NASSync/ld-decode.5.21.19/lddecode_core.py", line 1236, in __init__
    self.linelocs1, self.linebad, self.nextfieldoffset = self.compute_linelocs()
  File "/home/erik/NASSync/ld-decode.5.21.19/lddecode_core.py", line 970, in compute_linelocs
    line0loc, self.isFirstField = self.getLine0(validpulses)
  File "/home/erik/NASSync/ld-decode.5.21.19/lddecode_core.py", line 923, in getLine0
    line0loc_next, isNotFirstField = self.processVBlank(validpulses, 100)
  File "/home/erik/NASSync/ld-decode.5.21.19/lddecode_core.py", line 838, in processVBlank
    goodones = [vp[2] == True for vp in self.validpulses[firstblank:firstblank+24]]
TypeError: unsupported operand type(s) for +: 'NoneType' and 'int'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "ld-decode.py", line 109, in <module>
    print("current sample: " + ldd.fdoffset, file=sys.stderr)
TypeError: can only concatenate str (not "int") to str
Info: Processing complete
```

Will try skipping further ahead, I'll find a number that doesn't explode eventually.

happycube, 2019-06-11:
Try -s 80000 -S 1

simoninns, 2019-06-20:
Even if there is a work around; ld-decode shouldn't crash.  Relabelled and left open for fix @happycube

happycube, 2019-06-25:
I tweaked something in that area which *might* help... try again?

toynbeeidea, 2019-09-06:
Sorry, I don't have the .lds file anymore, deleted it due to lack of space.  Plus at the time I was capping both sides and telling ld-decode to ignore the endframe flag to attempt decoding both sides at once, which isn't supported.

closed 2019-09-06 by toynbeeidea