happycube/ld-decode #240: Include ld-decode build number in .json

Gamnn, 2019-06-15:
It would be useful to keep track of which specific ld-decode commit was used to decode a .tbc, perhaps in a comment at the top of a .json?

simoninns, 2019-06-15:
In the videoParameters would be the best place, as that is only included once per JSON file.  Then the info would be available to the tool chain too.

simoninns, 2019-07-14:
Related to #290 - the build number should be included in the ld-decode debug log file (changed issue title)

simoninns, 2020-01-28:
I've added the following to the top-level makefile:

`
### Get the current git commit information ###
BRANCH = $(shell git rev-parse --abbrev-ref HEAD)
COMMIT = $(shell git rev-parse --short HEAD)
`

These variables should be fine for importing into the Python stuff too @happycube and then added them to the JSON if desired.

The ld-decode-tools qmake .pro files have been altered to pull these in as defines (there is also a default in case compilation is local in Qt Creator (sets both to "unknown").

All tools will now report commit and branch using the --version switch.  The TBC library has also been altered so the commit and branch is included with the first debug message to pass through the library.

simoninns, 2020-01-28:
This is what it looks like from the tools:

[image]

Gamnn, 2020-01-28:
Just a thought, but a compatibility version would be useful too, for the .TBC json. This could help with any changes to the format (e.g. for #242) would break compatibility with old decodes.

happycube, 2020-10-18:
Trying to figure out how to copy this into a 'make install''d copy.  Any ideas?

happycube, 2022-01-17:
Finally got around to having "make install" stash the git version info into a string.  get_git_info() will use the file when the git commands do not work, so presumably it will work in an installed copy.

happycube, 2022-01-22:
I think this is fixed now with that commit...

closed 2022-01-22 by happycube