happycube/ld-decode #248: PAL CAV with EFM causes 2dB decrease in SNR

simoninns, 2019-06-18:
There is a clear correlation between the presence of EFM data and a 2dB drop in video signal quality (black level SNR) on all Domesday AIV discs.  The discs have sections with and without EFM data present.

For example; here is a partial decode of the EcoDisc side 1 which clearly shows the position of two sections of EFM (at the start and middle):

[image]

This is in contrast to side 2 of the same disc which is CAV as well, but does not contain any EFM data (and maintains a better video quality overall with a smooth expected curve of gradually increasing quality):

[image]

The same issue can be seen on the DD community discs which contain no EFM during the intro video (there is a short burst of EFM from the first frame, then the video with analogue audio, then EFM again.  At the end is a tutorial video with no EFM and analogue audio):

[image]

The same pattern can be seen on side 2 of the disc (which is almost identical in structure, although suffers edge-rot causing a more gradual SNR change at the very end of the disc):

[image]

It looks very likely that this bug is causing Domesday discs to have lower video quality than is actually possible.

simoninns, 2019-06-18:
By way of contrast and base-lining; here is the GGV1069 SNR (initial 15K frames) from the same capture setup

[image]

simoninns, 2019-06-18:
In case it's useful, here is a CSV of the VITS data from EcoDisc side 1:

[ecodisc_side1.tbc.zip](https://github.com/happycube/ld-decode/files/3299830/ecodisc_side1.tbc.zip)

simoninns, 2020-07-30:
Relabelled as a question...  Is the drop due to some sort of bug (unlikely) or is it just the effect the presence of EFM has on the overall noise floor (likely)?

happycube, 2020-07-31:
I think it's the effect on the noise floor.

simoninns, 2020-07-31:
alrighty then... let's close it up :)

closed 2020-07-31 by simoninns