happycube/ld-decode #252: PAL Crash during decode of Blackadder CLV

IanSmallshire, 2019-06-18:
python3 ld-decode.py /media/ian/LDS/blackadder/BlackAdder4_CLV_PAL_side3_2019-06-17_19-48-41.lds /media/ian/LDS/blackadder/BlackAdder4_CLV_PAL_side3 --pal -s 1 

[BlackAdder4_CLV_PAL_side3.cut.lds.zip](https://github.com/happycube/ld-decode/files/3302733/BlackAdder4_CLV_PAL_side3.cut.lds.zip)

file frame 3574 CLV timecode 2:13.03 frame 3328
file frame 3575 CLV timecode 2:13.04 frame 3329
file frame 3576 CLV timecode 2:13.05 frame 3330
file frame 3577 CLV timecode 2:13.06 frame 3331
file frame 3578 CLV timecode 2:13.07 frame 3332
file frame 3579 CLV timecode 2:13.08 frame 3333
file frame 3580 CLV timecode 2:13.09 frame 3334
file frame 3581 CLV timecode 2:13.10 frame 3335
ERROR - please paste the following into a bug report:
current sample: 5732120834
arguments: Namespace(MTF=None, MTF_offset=None, daa=False, ignoreleadout=False, infile='/media/ian/LDS/blackadder/BlackAdder4_CLV_PAL_side3_2019-06-17_19-48-41.lds', length=110000, nodod=False, noefm=False, ntsc=False, ntscj=False, outfile='/media/ian/LDS/blackadder/BlackAdder4_CLV_PAL_side3', pal=True, seek=-1, start=1, vbpf_high=None, vlpf=None)
Exception: unsupported operand type(s) for +: 'NoneType' and 'int'  Traceback:
  File "ld-decode.py", line 113, in <module>
    f = ldd.readfield()
  File "/home/ian/ldd-rev5/ld-decode/lddecode_core.py", line 2044, in readfield
    f, offset = self.decodefield()
  File "/home/ian/ldd-rev5/ld-decode/lddecode_core.py", line 2022, in decodefield
    f = self.FieldClass(self.rf, self.indata, self.rawdecode, audio_offset = self.audio_offset, prevfield = self.curfield)
  File "/home/ian/ldd-rev5/ld-decode/lddecode_core.py", line 1548, in __init__
    super(FieldPAL, self).__init__(*args, **kwargs)
  File "/home/ian/ldd-rev5/ld-decode/lddecode_core.py", line 1308, in __init__
    self.linelocs1, self.linebad, self.nextfieldoffset = self.compute_linelocs()
  File "/home/ian/ldd-rev5/ld-decode/lddecode_core.py", line 1038, in compute_linelocs
    blank2 = self.getblankrange(validpulses, blank1[1] + 1)
Info: Processing complete

confirmed on cut file

python3 ld-decode.py /media/ian/LDS/blackadder/BlackAdder4_CLV_PAL_side3.cut.lds /media/ian/LDS/blackadder/BlackAdder4_CLV_PAL_side3.cut --pal -s 1 -l 10
file frame 1 CLV timecode 2:13.08 frame 3333
file frame 2 CLV timecode 2:13.09 frame 3334
file frame 3 CLV timecode 2:13.10 frame 3335
ERROR - please paste the following into a bug report:
current sample: 7313678
arguments: Namespace(MTF=None, MTF_offset=None, daa=False, ignoreleadout=False, infile='/media/ian/LDS/blackadder/BlackAdder4_CLV_PAL_side3.cut.lds', length=10, nodod=False, noefm=False, ntsc=False, ntscj=False, outfile='/media/ian/LDS/blackadder/BlackAdder4_CLV_PAL_side3.cut', pal=True, seek=-1, start=1, vbpf_high=None, vlpf=None)
Exception: unsupported operand type(s) for +: 'NoneType' and 'int'  Traceback:
  File "ld-decode.py", line 113, in <module>
    f = ldd.readfield()
  File "/home/ian/ldd-rev5/ld-decode/lddecode_core.py", line 2044, in readfield
    f, offset = self.decodefield()
  File "/home/ian/ldd-rev5/ld-decode/lddecode_core.py", line 2022, in decodefield
    f = self.FieldClass(self.rf, self.indata, self.rawdecode, audio_offset = self.audio_offset, prevfield = self.curfield)
  File "/home/ian/ldd-rev5/ld-decode/lddecode_core.py", line 1548, in __init__
    super(FieldPAL, self).__init__(*args, **kwargs)
  File "/home/ian/ldd-rev5/ld-decode/lddecode_core.py", line 1308, in __init__
    self.linelocs1, self.linebad, self.nextfieldoffset = self.compute_linelocs()
  File "/home/ian/ldd-rev5/ld-decode/lddecode_core.py", line 1038, in compute_linelocs
    blank2 = self.getblankrange(validpulses, blank1[1] + 1)
Info: Processing complete

IanSmallshire, 2019-06-18:
Second issue of the TBC failing
[image]

File on my google drive as could no recreate with 10 frames
https://drive.google.com/open?id=1TBtqOuUC2EwyzAfS7IeToXLr-CjFFC-D

simoninns, 2019-06-18:
The 13,000 dot long drop outs on each field should be noted @happycube

IanSmallshire, 2019-06-18:
Showing where the dropouts are marked
[image]

happycube, 2019-06-25:
still a mess, but removed the crash

IanSmallshire, 2019-06-25:
Have confirmed that the crash does not occur during processing the full file. Currently at 5K frames

happycube, 2019-06-27:
Disk frequency levels are too low - black level at -5 "IRE" white at 90.

happycube, 2019-06-27:
Taking care of #218 will fix the remaining issues (mass dropout, vrooping), so closing this issue.

closed 2019-06-27 by happycube