happycube/ld-decode #257: PAL field order and .tbc format

skunge2000, 2019-06-22:
A question about the .tbc format and PAL 4fsc sampled video from a colleague.

It appears as if the .tbc files consist of 626-line 'frames' (the last line being a duplicate of the first line in the next frame) but when I look at the contents here it's the second field in each frame which starts with a half-line. That line (line 23) should be in the first field, surely?

In PAL 625 standard video, active video starts in the second half of line 23 (which is the first active video line in field 0) ends at line 310 (which is a full line) then starts again on line 336 (which is the first active video in field 1 and a full line) and ends with the first half of line 623 (a half line of active video).

Therefore the first field, field 0 should start with a half-line of active video and end with a full line, the second field, field 1, should start with a full-line of active video and end with a half line.  The .tbc file appears to be the opposite of this?

Are the fields out-of-phase (i.e. field 1 from frame N is followed by field 0 from frame N+1) ? or temporally reversed (i.e. field 1 from frame N if followed by field 0 from frame N).

Is this an 'oddity' with PAL Laserdisc (a bit like PAL DV25 that has reverse field dominance to usual?)  Or am I/are we missing something ?

skunge2000, 2019-06-22:
Additional images for reference here : 
[image]
[image]

happycube, 2019-06-22:
That's a *nice* table, I wish I had that earlier.  Is there one for 480i as well? ;)

Looks like I might have the field order reversed - right now I'm fixing crashing/skipping issues for rev5.2, so I'll dig into this for rev6, since reversing it now would confuse the existing tools.

rtrussell, 2019-06-22:
> Looks like I might have the field order reversed 

Can we at least be confident that the _temporal_ order of the fields is correct, in other words irrespective of which is 'field 1' and which 'field 2' that the lines in the TBC file should be presented (on a conventional interlaced display) in the order in which they appear in the file?

happycube, 2019-06-22:
Definitely.

skunge2000, 2019-06-22:
Here's the 480i version :
[image]
[image]

simoninns, 2019-06-22:
Just ordered a copy from ebay; @happycube - there are some good copies state-side too for a few bucks.

skunge2000, 2019-06-22:
Second edition is a bit more abridged when it comes to composite. (I now have both...)

simoninns, 2019-11-11:
I think this question is answered now; so closing issue.  Please re-open with comment if anyone disagrees.

closed 2019-11-11 by simoninns