happycube/ld-decode #258: PAL decode field ordering inconsistent

simoninns, 2019-06-23:
When decoding the same disc from two different captures, the PAL field ordering is inconsistent.

Decodes are the same disc, captured on two different players (DomesdayDuplicator1 and DomesdayDuplicator2 - both are LD-V4300Ds).

Command line for duplicator1:

[image]

Command line for duplicator2:

[image]

ld-analyse result for duplicator1:

[image]

ld-analyse result for duplicator2:

[image]

Both discs are Domesday National Side 1 (disc set 2)

skunge2000, 2019-06-23:
Could this be related to https://github.com/happycube/ld-decode/issues/257#issue-459481921

simoninns, 2019-06-23:
Yes it could be; but I only found the regression this morning; so @happycube needs a little time to look into the cause (and in what commit it occurred since it might be very recent as I test with the on-going rev6 commits).

simoninns, 2019-06-23:
The issue has been analysed and the fault seems to be within the ld-decode-tools shared library - the output from ld-decode is consistent between the decodes and the .tbc (and .json) are correct.  I need to look into the tools library and see why this issue is occurring.

happycube, 2019-06-23:
On the ld-decode end, I just put up a block so that the .tbc file will only start with a first field (important for rev5.2, which will ship with rev5.1 tools)

simoninns, 2019-06-23:
Issue appears to be a GUI error in ld-analyse which is transposing the VBI data into the wrong field:

[image]

simoninns, 2019-06-23:
Cleaned up ld-analyse VBI handling and fixed bug that occurred in the shared-library that incorrectly reloaded certain metadata when a GUI application loaded a second file without restart.

closed 2019-06-23 by simoninns