happycube/ld-decode #263: Merge 2D NTSC comb filter code with ld-comb-pal

simoninns, 2019-06-27:
Since the 3D code is not functioning for NTSC and not present for PAL, it would be a good idea to merge the 2D comb-filters into a single code base (the 3D could then be added as a common function to both filters).  This would involve merging the 2D NTSC core filter into the PAL front-end (which is far more advanced than the current NTSC code and supports multithreading).

simoninns, 2019-06-27:
Related issues #211, #133, #126, #123, #119, #104, #93

atsampson, 2019-06-28:
This would also be a good opportunity for a more generic name -- ld-chroma-decoder? The current ld-comb-pal doesn't actually use a comb filter; it uses a pair of raised-cosine bandpass filters.

simoninns, 2019-06-28:
This is very true; the name 'comb-filter' was due to the previous implementation and that it is a commonly accepted name for a chroma decoder in the NTSC universe.  I like the idea of renaming it to ld-chroma-decoder.

simoninns, 2019-07-11:
Complete

PR #280 (2019-07-02): Merge ld-comb-ntsc and ld-comb-pal into ld-chroma-decoder
The new tool has three backends, chosen with `-f`:

- pal2d - equivalent to ld-comb-pal
- ntsc2d - equivalent to ld-comb-ntsc
- ntsc3d - equivalent to ld-comb-ntsc --3d

They all use ld-comb-pal's thread pool; pal2d and ntsc2d will run in parallel, and ntsc3d will only use one thread for now (since each frame depends on the previous one). By default, it'll use either ntsc2d or pal2d depending on the input metadata, so you can usually replace existing calls to ld-comb-{pal,ntsc} with ld-chroma-decoder.

I've tested this with various samp [... cut]
closed 2019-07-11 by simoninns