happycube/ld-decode #272: PAL/NTSC 3-way vote correction through combination

simoninns, 2019-06-28:
If you have 3 copies of the same disc you could process it like aeroplanes do.  Each copy supplies a pixel into a 3-way vote; if all 3 agree, take the median; if only 2 agree discard the 3rd and take the median

It would magically remove bad data from DOs and perform a median stack at the same time and it's simple(ish) to implement at the PAL/NTSC point.

If all 3 disagree, it could be flagged as a DO into the metadata - then you'd also know automatically where it didn't work and you could DOC the result as a final step (the possible DOD back into the 3 copies would likely out perform anything ld-decode can do by RF analysis).  Another advantage is that the source format doesn't matter at all; it would work as good on pull-down as anything else.

simoninns, 2019-06-28:
As an additional note; if the voting disregards sync data, it could also only act on NTSC/PAL data in the expected 0IRE - 100IRE range; furthering the resistance of the technique to bad data.

atsampson, 2019-06-28:
In statistical terms, you have a set of measurements with N=3, and you're asking whether discarding outliers before computing the median would give you a better estimate of the actual value. [Anscombe and Barron (1966), "Treatment of Outliers in Samples of Size Three"](https://nvlpubs.nist.gov/nistpubs/jres/70B/jresv70Bn2p141_A1b.pdf) says it should do.

simoninns, 2019-06-28:
Thanks for the additional reference; that's basically the lines I was thinking along; however, it occurs to me that, if you remove the outliers, then averaging rather than median might be a better way to treat the remaining measurements - as averaging would assist in the reduction of noise.  However, I haven't spent any real time looking into it, so it's open to suggestion and experimentation at the moment.

nobananasforyou, 2019-06-28:
https://github.com/dubhater/vapoursynth-median
Is this helpful?

simoninns, 2019-06-29:
Thanks, I'm aware that there are median stacking applications for video (and I will look into them further) - however the intention with this issue is not to process video, rather NTSC and PAL before information is lost in the transformation into video via chroma-decoding.

simoninns, 2019-08-18:
Implemented in ld-combine and ready for test

closed 2019-08-18 by simoninns