happycube/ld-decode #277: Possibly incorrect C2 CIRC error correction

agren, 2019-07-01:
I have been browsing the code and found something that doesn't add up in [c2circ.cpp](https://github.com/happycube/ld-decode/blob/3d5cdeddbaf4515b41d0dff2c43faa51d649c132/tools/ld-process-efm/c2circ.cpp) (current master).

It appears to me that rs.decode() take input on the format `data ++ Q parity`. While `C2Circ::errorCorrect(void)` send in data with the format `data[0:12] ++ Q parity ++ data[12:24]`.

So I would have expected that the `// Convert the data and errors into the form expected by the ezpwd library` step would be more like:
```
    ...
    data[0] = static_cast<uchar>(interleavedC2Data[0]);
    data[1] = static_cast<uchar>(interleavedC2Data[1]);
    data[2] = static_cast<uchar>(interleavedC2Data[2]);
    data[3] = static_cast<uchar>(interleavedC2Data[3]);
    data[4] = static_cast<uchar>(interleavedC2Data[4]);
    data[5] = static_cast<uchar>(interleavedC2Data[5]);
    data[6] = static_cast<uchar>(interleavedC2Data[6]);
    data[7] = static_cast<uchar>(interleavedC2Data[7]);
    data[8] = static_cast<uchar>(interleavedC2Data[8]);
    data[9] = static_cast<uchar>(interleavedC2Data[9]);
    data[10] = static_cast<uchar>(interleavedC2Data[10]);
    data[11] = static_cast<uchar>(interleavedC2Data[11]);

    data[12] = static_cast<uchar>(interleavedC2Data[16]);
    data[13] = static_cast<uchar>(interleavedC2Data[17]);
    data[14] = static_cast<uchar>(interleavedC2Data[18]);
    data[15] = static_cast<uchar>(interleavedC2Data[19]);
    data[16] = static_cast<uchar>(interleavedC2Data[20]);
    data[17] = static_cast<uchar>(interleavedC2Data[21]);
    data[18] = static_cast<uchar>(interleavedC2Data[22]);
    data[19] = static_cast<uchar>(interleavedC2Data[23]);
    data[20] = static_cast<uchar>(interleavedC2Data[24]);
    data[21] = static_cast<uchar>(interleavedC2Data[25]);
    data[22] = static_cast<uchar>(interleavedC2Data[26]);
    data[23] = static_cast<uchar>(interleavedC2Data[27]);

    data[24] = static_cast<uchar>(interleavedC2Data[12]);
    data[25] = static_cast<uchar>(interleavedC2Data[13]);
    data[26] = static_cast<uchar>(interleavedC2Data[14]);
    data[27] = static_cast<uchar>(interleavedC2Data[15]);
    ...
```
And then the convert it back after rs.decode() has been run.

It might be that I am missing something. My means to test are limited since I don't have a proper EFM stream to experiment with.

simoninns, 2019-07-02:
I'm almost sure that I had the code that way when I first wrote it, but the error correction didn't work.  As far as I remember there was no requirement for the parity symbols to be at the end of the correction word.  I will give it a recheck though (but right now the C2 ERC is functioning for both error detection and correction at the C2 level).

simoninns, 2019-07-08:
Since I was looking into the error correction today, I decided to verify this.  The C2 error correction does not function if you move the parity symbols to the end of the input (as per your example).  I can't say I am super-knowledgable about such CRC, but one way it works and the other it doesn't :)  I will leave the code as-is.  If you have any more info, or if you have managed to alter the code successfully, please reopen this issue and comment. (also, thanks for taking the time to review the code!)

closed 2019-07-08 by simoninns

agren, 2019-07-08:
Yeah, if it works it's probably correct :) I must be missing something.
I've managed to sample the EFM-signal from a CD-player so I'll have a go with that.

simoninns, 2019-07-09:
If you are wanting to use the decoder for something; the main development of it is in the rev6 branch - that's where I am (currently) coding the production version.  It has some major fixes over the original code and should be in beta condition really soon