happycube/ld-decode #282: Remove (or make optional) cropping in ld-chroma-decoder

simoninns, 2019-07-03:
Cropping of the PAL/NTSC visible area is really a matter of taste (and well supported by external video processing tools) - it should be removed from the ld-decode-tools - non-visible areas of PAL and NTSC should simply be replaced with solid black.

Gamnn, 2019-07-03:
Options to leave it uncropped and unmasked could be useful, but I vote the crop remains by default. The full dimensions aren't evenly divisible by 8, so it would cause trouble with many video toolchains.

simoninns, 2019-07-03:
I'd counter that the ld-decode-tool chain's purpose is to process PAL and NTSC; there's an argument for leaving in the non-visible information too (especially VBI).  It would be more 'preservation' and less 'lossy' for the chroma-decoder to simply convert non-visible picture dots to RGB grey (adjusting for expected IRE levels) and output the whole frame.

Cropping on the RGB/YUV level is, strictly speaking, outside of the scope of the chroma-decoder (it was originally done to get the .rgb file size down - but that issue was removed by pipelining support).

Gamnn, 2019-07-03:
Unless padded so that they're mod8 or 16, the pipelining support won't help as much- it'll be slower to process as they're not aligned.

I think the chroma-decoder tool should be where it prepares it for video processing. The VBI and such is better preserved in their native fields in the .tbcs, rather than converted to frames after chroma-decoder.

simoninns, 2019-07-03:
Again, this is specialisation for ffmpeg - nothing to do with chroma decoding.  Just because ffmpeg can't use the VBI, doesn't mean other tools (like teletext converters) can't - and those would probably expect RGB/YUV video (rather than PAL/NTSC).

simoninns, 2019-07-03:
As another note; making lots of stuff optional exponentially increases the issue of maintenance and testing - we got into that mess with ld-decode and had to spend a lot of effort unwinding it.  For the ffmpeg specialisation another (dedicated) tool would be architecturally more sensible.

atsampson, 2019-07-03:
I also think it makes sense to keep the cropping in ld-chroma-decoder -- the decoder and output writer have to operate within *some* bounds, and it'd be straightforward to make that be either the active region (extracted from the metadata) or the whole frame if an option's specified. Having ld-chroma-decoder always process the whole frame would slow it down considerably, and using a separate cropping tool would mean some additional complexity to pass the bounds from the metadata to it.

A caveat with whole-frame output is that the 2D decoders need to look at surrounding lines, so we might need some padding around the input data.

atsampson, 2019-07-03:
On the other hand, padding to a convenient size and YUV-to-RGB conversion are both things that ffmpeg can do itself...

simoninns, 2019-07-03:
Firstly, let's drop the YUV to RGB from this issue (it is already covered in another issue iirc).

On the cropping; I don't think it would be too much of an overhead - the chroma filter should only filter the active area (as it does at the moment).  Really my idea here is that the inactive area is converted from the grey16 to the same output format as the active area (as the non-active part of a frame doesn't contain chroma data); the active area is then overlaid on the inactive for output.  This would maintain the frame dimensions whilst  including the non-active picture information for tools which aren't specialised for ld-decode's PAL/NTSC grey16 format.

simoninns, 2019-07-04:
It was decided (via IRC) that this isn't required

closed 2019-07-04 by simoninns