happycube/ld-decode #283: ld-chroma-decoder should output yuv natively

Gamnn, 2019-07-03:
Rather than converting the output to RGB48, yuv444 would be preferable for most toolchains.
RGB could be left in as an option.

Piggybacking on this issue, an additional option to output with yuv4mpeg headers would be useful for piping into ffmpeg and other tools.

simoninns, 2019-07-03:
YUV output - yes; specialisation of the output for ffmpeg; probably not.  As I said on IRC both metadata stream combination and 'prep' for ffmpeg would be better served by another tool.  It has nothing to do with chroma decoding.

Gamnn, 2019-08-20:
#119 can be handled by downstream tools when this is implemented, as it will keep luma and chroma channels separate.

atsampson, 2021-02-15:
Implemented in #618.

closed 2021-02-15 by atsampson