happycube/ld-decode #284: Support VBI output as per the linuxtv raw VBI format

simoninns, 2019-07-04:
See: https://linuxtv.org/downloads/v4l-dvb-apis/uapi/v4l/dev-raw-vbi.html#raw-vbi

Unless anyone knows of a more suitable output format for external tools?

simoninns, 2019-07-04:
There is also a 'sliced' format: https://linuxtv.org/downloads/v4l-dvb-apis/uapi/v4l/dev-sliced-vbi.html

simoninns, 2019-08-19:
This is probably best implemented as part of the new ld-discmap tool as that 'preps' the TBC for final output

Gamnn, 2020-01-08:
Sounds like a job for ld-export-metadata.

simoninns, 2020-01-08:
Working from the JSON metadata won't output any unimplemented VBI lines though (like teletext for example).  I think the linuxtv format requires the actual video lines rather than decoded LD-specific VBI data points.

Gamnn, 2020-01-08:
Ah, that's true.

I'm uncertain if ld-discmap is the right place for it either, though. At least until it can at minimum pass-through weird ntsc pull-down tbcs if not fully support them. I believe Cinderella PILA-1124 might be one such case, but I'd have to check.

simoninns, 2020-01-08:
ld-discmap isn't the right place for it - I think a separate tool would be required to pull the lines from the .tbc and output them in the right format for linuxtv.

atsampson, 2020-01-08:
I've been experimenting with [libzvbi](http://zapping.sourceforge.net/ZVBI/), which includes both a decoder that extracts various kinds of ("sliced") VBI data from raw images, and pretty complete Teletext and CC renderers.

At present I have the zvbi decoder hooked up in ld-process-vbi, alongside Simon's existing CC decoder, and it seems to extract both PAL Teletext and NTSC CC frames successfully. The next step would be to make it store the "sliced" results in the JSON (as with CC already), then have ld-export-metadata extract it again, feed it into the renderer, and write the result out as subtitle files (or just dump the sliced data out if you want to render the whole magazine with an external tool).

atsampson, 2020-01-08:
(That's not to say that this is necessarily the right way to do it in the long run, but I want to see if it works at all before figuring out the right way to do it!)

simoninns, 2020-01-08:
If you are happy to code it, you'll get no complaints from me :)  Feel free to experiment - it's better for us to discuss results rather than potential results!

atsampson, 2020-01-09:
Work in progress on the [zvbi](https://github.com/atsampson/ld-decode/tree/zvbi) branch - ld-export-metadata can now write out Teletext lines in formats that ZVBI's tools and vhs-teletext's tools can work with:

[image]

I'm pretty convinced now that going via the JSON is *not* the right way to handle Teletext, though - it works, but it produces very big JSON files (which JsonWax doesn't handle well), and getting it into and out of the metadata is half the effort...

atsampson, 2020-01-17:
Worth noting that there's some variation between discs in which lines are used. Using ld-analyse's line numbering system, [British Garden Birds](https://www.lddb.com/laserdisc/39164/BBCV-1005L/British-Garden-Birds) and [Touch of Evil](https://www.lddb.com/laserdisc/29742/PLFMB-32511/Touch-of-Evil) have Teletext on lines 39-42 (two lines per field), but [Glenn Gould's Greatest Hits](https://www.lddb.com/laserdisc/58502/SLV48433/Glenn-Gould:-Greatest-Hits) has Teletext on lines 27-30 *and* 39-42 (four lines per field).

atsampson, 2022-12-24:
vhs-teletext can now read .tbc files, so I don't know if this is still needed...

harrypm, 2025-11-30:
Closing as now redundantly solved.

- vhs-teletext supports `.tbc.
- FFmpeg can be made to do a luma only export of the VBI space in grayscale.
- The chroma decoder can be made to just export the VBI space with active area commands.
- IMX style 720x512/720x608 export is also now a tbc-video-export profile.