happycube/ld-decode #289: ld-decode JSON errors

simoninns, 2019-07-13:
ld-decode is producing two undocumented (and unwanted) fields in the JSON:

decodeFaults
discLoc

These need to be removed (or justified - which downstream tool requires these?)

ld-decode is also generating a dropOuts object (and 3 associated array elements) for every field (even though there are no drop-outs and all elements are empty).  dropOuts is an optional object and should only be included if required by the field (as documented by the JSON metadata format wiki).

happycube, 2019-07-13:
decodeFaults is a bitmap for certain error conditions, and discLoc is there to tell me where to start another decode from if something goes odd... I'll put the latter in verboseVITS, and I'll see about the dropouts.

simoninns, 2019-07-14:
I'll close this issue now the dropouts are fixed, and raise a new one for the missing debug log

closed 2019-07-14 by simoninns