happycube/ld-decode #290: ld-decode requires a debug log file output for each decode

simoninns, 2019-07-14:
Right now debug information is being pushed into JSON metadata (which is for tool communication) or debug is only appearing on the console (which, due to the length of the decoding process, is rarely seen).

ld-decode should produce a log file for each decode using the .lds filename as a base.  The debug file should contain:

A detailed list of all command line parameters
Details of the input .lds file (perhaps even the file size or an md5 hash)
The start and end datetime of the decode
One or more rows for each field processed containing the location on disc, any decoding issues and stats about each field as appropriate.

Each line of the debug should be time-stamped to assist with performance analysis.

Wherever possible, the debug should be verbose (rather than a series of numbers, boolean values with no labelling).

As part of this issue, the remaining debug in the JSON (and the --verboseVITS option) should be removed.

simoninns, 2020-08-13:
Just want to bump this issue as a reminder... Now we are getting to the stage where many discs are decoded using automatic scripts - finding bugs and failures is getting harder and harder without any logging.

happycube, 2020-12-13:
Thinking about this, I'm going to make a 'logger' sub-class and hook what existing logging there is into this file, and then go from there.  Shouldn't be too bad so not bumping to rev8 along with #577.

simoninns, 2020-12-29:
Looks good in master.  Can this be closed @happycube ?