happycube/ld-decode #293: Change analogue PCM audio sample rate to 44.1KHz

simoninns, 2019-07-14:
The sample rate of analogue PCM is currently 48KHz.  This should be switched to 44.1KHz to match EFM digital audio, cut down the file size and make sound sync easier.

Gamnn, 2019-10-28:
I'd personally rather it be left at 48kHz as it's the de-facto standard for digital video formats, and 44.1kHz isn't as widely supported (DVD-Video can only be 48kHz or 96kHz for example). When 44.1kHz is needed, it can be downsampled after decode, and would be preferable to upsampling.

happycube, 2019-12-01:
Making 44.1khz the default for consistency and ease of refining CX filter.  It will be trivial to add a command line option to use any audio rate you want though :)

closed 2019-12-01 by happycube