happycube/ld-decode #318: NTSC 3D artefacts on shot changes

atsampson, 2019-08-18:
The NTSC 3D decoder seems to produce incorrect output when the two fields it's looking at are significantly different.

For example, frame 115 of GGV1069 -- the middle frame in the crosstalk test -- should look like this, and does with ntsc2d:

[image]

But it looks like this with ntsc3d:

[image]

Patching Comb to always select the 3D decoder gives this:

[image]

I tried going back to rev4 and the 3D output is the same (we just weren't seeing it because the motion detection wasn't using it), so it doesn't look like a recent breakage.

simoninns, 2019-08-19:
Moved to rev7 with the other 3D chroma-filter issues.

happycube, 2020-07-07:
ntsc3d is de-implemented right now.  This will probably come back when it's re-implemented but it'll be different and another story... er, issue.

closed 2020-07-07 by happycube