happycube/ld-decode #325: #234 still not quite fixed in pre-release 5.2

toynbeeidea, 2019-09-06:
Still getting the same problems as #243 - but a lot less than 5.1 yielded.  I'm guessing it just needs further tweaking?

>  file frame 2142 CLV timecode 1:08.20 frame 2060
> file frame 2143 CLV timecode 1:08.21 frame 2061
> WARNING:root:NTSC field phaseID sequence mismatch (player may be paused)
> file frame 2144 CLV timecode 1:08.22 frame 2062
> WARNING:root:NTSC field phaseID sequence mismatch (player may be paused)
> file frame 2145 CLV timecode 1:08.23 frame 2063
> file frame 2146 CLV timecode 1:08.24 frame 2064
> file frame 2147 CLV timecode 1:08.25 frame 2065

> file frame 2176 CLV timecode 1:09.24 frame 2094
> file frame 2177 CLV timecode 1:09.25 frame 2095
> WARNING:root:NTSC field phaseID sequence mismatch (player may be paused)
> file frame 2178 CLV timecode 1:09.26 frame 2096
> WARNING:root:NTSC field phaseID sequence mismatch (player may be paused)
> file frame 2179 CLV timecode 1:09.27 frame 2097
> file frame 2180 CLV timecode 1:09.28 frame 2098

[https://drive.google.com/open?id=19-knGkegMFTsqzM7H9-tzVQnk6E4ZcdI](url)

Gamnn, 2019-09-08:
First one but not the second reproduces in rev6.

```

file frame 14 CLV timecode 1:08.20 frame 2060
file frame 15 CLV timecode 1:08.21 frame 2061
WARNING:root:NTSC field phaseID sequence mismatch (player may be paused)
file frame 16 CLV timecode 1:08.22 frame 2062
WARNING:root:NTSC field phaseID sequence mismatch (player may be paused)
file frame 17 CLV timecode 1:08.23 frame 2063

```
[image]

Possibly related to #324

simoninns, 2019-11-11:
There have been a number of improvements to ld-decode that should fix this type of issue.  Suggest retest with latest rev6 commit and re-open if the problem still exists.

simoninns, 2019-11-19:
Reopening as issue is reported as still present

toynbeeidea, 2019-12-11:
This part, at least, appears to be fixed in #351.

closed 2019-12-11 by toynbeeidea