happycube/ld-decode #326: Tool to process VITS

Gamnn, 2019-09-11:
Currently, VITS is only processed in ld-decode.py. I propose either ld-process-vbi or a new tool be written to process VITS on a .tbc, which would be helpful for comparing the results of things like ld-combine on SNR.

happycube, 2019-09-11:
How to do SNR:

- Convert output to IRE level (0-100)
- signal:
  - SNR: mean of IRE-level output
  - PSNR: 100
- noise: standard deviation of IRE-level output

(P)SNR: 20 * log10(signal/noise)

happycube, 2019-09-11:
Long expired patent about VITS analysis by Discovision Associates:

https://patents.google.com/patent/US4455634A/en?oq=4455634

atsampson, 2019-12-24:
This could also be used to automatically adjust the black/white levels, which are off for a lot of sources at the moment - e.g. here's ve-snw-cut.lds from testdata:

[image]

You can also see some HF attenuation there (the baseline of the curved pulse should be flat on the 0 IRE line), which will mess up the amplitude of the chroma signal. At the moment ld-chroma-decoder has a fixed fudge factor for this - if we could measure the VITS, we could do proper video EQ instead.

Gamnn, 2020-01-24:
#218 should help with that, but yeah, it would be useful to have EQ at the chroma-decoder stage as well.

atsampson-abertay, 2020-01-24:
There's an argument for doing both - EQ on the RF to compensate for disc, player and ADC nonlinearity (i.e. a better-informed version of the MTF compensation filter ld-decode already has), and EQ on the .tbc data to compensate for errors in the video source. I've got discs where the VBI looks fine, but the video itself has obvious phase/EQ problems (so the fault was in the original source, not the Laserdisc mastering).

atsampson, 2020-04-16:
For testing purposes, [American Graffiti (1973) [42726]](https://www.lddb.com/laserdisc/07157/42726/American-Graffiti) is an example of a disk that _doesn't_ have VITS:

[image]

simoninns, 2020-09-08:
@happycube - for the current SNR (white/black) measurements performed by ld-decode... can you specify which VITS is being used (and if possible which sections of the VITS) for both PAL and NTSC?  I'd like this to match as closely as possible with the existing implementation.

simoninns, 2020-09-11:
From Discord:

happycube: the code that does it is in lddecode.py lines ~3042-3158.  there are several potential line slices it looks at, and the chosen slice could probably be added to the json
happycube: for pal the white area is always line 19, 12-20usec
happycube: and PAL IEC has a nice black area - line 22 after the burst, it looks like
happycube: PAL VITS areas are consistent at least.  NTSC white can be chosen from 3 different areas, depending on field/disk mastering

simoninns, 2020-09-14:
There is no lddecode.py.  Code is from:

https://github.com/happycube/ld-decode/blob/master/lddecode/core.py#L3041
to
https://github.com/happycube/ld-decode/blob/master/lddecode/core.py#L3157

happycube, 2020-09-14:
Doh, there's a reason I didn't copy+paste it in myself.  That's what I meant.

simoninns, 2020-09-16:
ld-process-vits tool added to master.  Recalculates the white and black SNR (doesn't do any of the extended VITS analysis provided by ld-decode), but it's fit for purpose.

closed 2020-09-16 by simoninns