happycube/ld-decode #332: VBI frame numbers on A-ko

Gamnn, 2019-09-17:
There's an odd pattern to the frame numbers on the first A-ko movie, on both the original and re-release. It comes out something like this, both in ld-decode and ld-process-vbi:

```
file frame 18521 unknown
file frame 18522 CAV frame 16665
file frame 18523 CAV frame 16663
file frame 18524 CAV frame 14765
file frame 18525 CAV frame 14766
file frame 18526 unknown
file frame 18527 CAV frame 16665
file frame 18528 CAV frame 16663
file frame 18529 CAV frame 14769
file frame 18530 CAV frame 14770
file frame 18531 unknown
```

Scrolling through on a player shows the proper frame numbers, without the odd non-sequential numbers after each pulldown frame (unknown).

[sample](https://drive.google.com/open?id=1-LlbBP_fjlF3eLUHpp5LZpOxmz_XoGRD)

atsampson, 2019-09-17:
In the frames that report incorrect numbers, VBI line 17 has a mangled value, but line 18 is correct:

```
Info: decoded 17= 16335365 18= 16335365 to picno= 14205
Info: decoded 17= 16335366 18= 16335366 to picno= 14206
Info: decoded 17= 16343007 18= 16335367 to picno= 16645
Info: decoded 17= 16343005 18= 16335368 to picno= 16643
Info: decoded 17= 16335369 18= 16335369 to picno= 14209
Info: decoded 17= 16335376 18= 16335376 to picno= 14210
Info: decoded 17= 16343005 18= 16335377 to picno= 16643
Info: decoded 17= 16343007 18= 16335378 to picno= 16645
```

e.g. line 17 for picno 14212 has the nonsense value 0xf95fdf, rather than 0xf94212.

A reasonably simple fix in `vbidecoder.cpp` would be to check that the BCD digits are actually in the range 0-9 before trusting the value -- this would also help with dropouts in the VBI.

simoninns, 2019-09-17:
Good idea @atsampson - also the fallback for a failed VBI code could also be to try to use line 18 instead (since the check condition would be there anyway).

simoninns, 2019-09-18:
Fixed

Gamnn, 2019-12-31:
It's still a little off, but only in the VBI frame number ranges of 9000-9999, 19000-19999, and 28000-28999 where it reports them as early CLV timecodes.

```
file frame 35068 CAV frame 27996
file frame 35069 CAV frame 27997
file frame 35070 CAV frame 27998
file frame 35071 CAV frame 27999
file frame 35072 unknown
file frame 35073 early-CLV minute 743
file frame 35074 early-CLV minute 743
file frame 35075 CAV frame 28002
file frame 35076 CAV frame 28003
file frame 35077 unknown
file frame 35078 early-CLV minute 743
file frame 35079 early-CLV minute 743
file frame 35080 CAV frame 28006
file frame 35081 CAV frame 28007
```

[9000.zip](https://github.com/happycube/ld-decode/files/4012063/9000.zip)
[19000.zip](https://github.com/happycube/ld-decode/files/4012064/19000.zip)
[28000.zip](https://github.com/happycube/ld-decode/files/4012065/28000.zip)

It appears to only be broken in ld-decode.
ld-process-vbi adds the correct CAV framenumber, but does not remove the erroneous CLV timecode.

simoninns, 2020-01-01:
Updated labels as this isn't an ld-decode-tools issue

closed 2020-01-05 by happycube

Gamnn, 2020-01-13:
Still an issue in ld-decode on ranges 38000 through 39999.

ld-analyse displays the correct frame numbers even without running it through ld-process-vbi, so seems to mainly be a display issue?

[38000.zip](https://github.com/happycube/ld-decode/files/4051869/38000.zip)
[39000.zip](https://github.com/happycube/ld-decode/files/4051870/39000.zip)

Gamnn, 2020-01-13:
scratch that- was on the wrong commit.

PR #400 (2020-01-05): Three VBI fixes for ld-decode
If seeking with `-S` fails, exit 1 rather than continuing from the wrong position.

Recognise the lead-in VBI code, rather than showing it as unknown.

Ignore VBI codes containing invalid BCD digits. Fixes #332.