happycube/ld-decode #336: ld-process-efm pads incorrectly on disc with odd initial disc time.

Gamnn, 2019-09-20:
```
ld-process-efm --debug
Debug: Configuration::readConfiguration(): Reading configuration from "/home/gamn/.config/ld-process-efm.ini"
Debug: Configuration::readConfiguration(): sourceDirectory = "/media/gamn/SSD/PCLP00006-Erik"
Debug: Configuration::readConfiguration(): audioDirectory = "/media/gamn/SSD/PCLP00006-Erik"
Debug: Configuration::readConfiguration(): dataDirectory = "/home/gamn"
Debug: MainWindow::on_actionOpen_EFM_file_triggered(): Called
Gtk-Message: 23:40:19.189: GtkDialog mapped without a transient parent. This is discouraged.
Debug: MainWindow::on_actionOpen_EFM_file_triggered(): Setting EFM source directory to: "/media/gamn/SSD/PCLP00006-Erik"
Debug: Configuration::writeConfiguration(): sourceDirectory = "/media/gamn/SSD/PCLP00006-Erik"
Debug: Configuration::writeConfiguration(): audioDirectory = "/media/gamn/SSD/PCLP00006-Erik"
Debug: Configuration::writeConfiguration(): dataDirectory = "/home/gamn"
Debug: Configuration::writeConfiguration(): Writing configuration to disk
Debug: MainWindow::on_actionOpen_TBC_file_triggered(): Set current file name to to: "/media/gamn/SSD/PCLP00006-Erik/side2-erik.efm"
Debug: MainWindow::on_decodePushButton_clicked(): Called
Debug: MainWindow::on_decodePushButton_clicked(): Opened EFM input file
Debug: MainWindow::on_decodePushButton_clicked(): Opened audio output temporary file
Debug: MainWindow::on_decodePushButton_clicked(): Opened data output temporary file
Debug: EfmProcess::setAudioErrorTreatment(): Audio error treatment set to conceal
Debug: EfmProcess::setAudioErrorTreatment(): Conceal type set to linear interpolation
Debug: EfmProcess::setDecoderOptions(): Pad initial disc time is false
Debug: EfmProcess::setDecoderOptions(): Decode as audio is true
Debug: EfmProcess::setDecoderOptions(): Decode as data is false
Debug: EfmProcess::run(): Thread initial run
Debug: EfmProcess::run(): Thread wake up on restart
Debug: EfmProcess::run(): Thread sleeping pending restart
Debug: MainWindow::processingCompleteSignalHandler(): Processing complete - audio available
Info: 
Info: EFM to F3 Frames:
Info:             Valid syncs: 12408892
Info:         Overshoot syncs: 23824
Info:        Undershoot syncs: 8967
Info:             TOTAL syncs: 12441683
Info: 
Info:       Valid EFM symbols: 410204395
Info:     Invalid EFM symbols: 163121
Info:          EFM error rate: 0.03975%
Info: 
Info:       In range T-values: 1516495374
Info:   Out of range T-values: 0
Info:          TOTAL T-values: 1516495374
Info: 
Info:            Valid frames: 12400682
Info:        Overshoot frames: 20537
Info:       Undershoot frames: 20465
Info:            TOTAL frames: 12441684
Info: 
Info: F3 Frame synchronisation:
Info:    Total input F3 Frames: 12441684
Info:         Discarded Frames: 109397
Info:     Total valid sections: 125583 ( 12307134 F3 Frames )
Info: 
Info: F3 Frame to F2 Frame decode:
Info:       Total input F3 Frames: 12307134
Info:      Total output F2 Frames: 12306840
Info:         Total Preemp Frames: 0
Info:   F3 Sequence Interruptions: 1
Info:           Missing F3 Frames: 134456
Info:           Initial disc time: 18:14.43
Info:             Final disc time: 46:27.21
Info: 
Info: F3 to F2 frame C1 Error correction:
Info:   Total C1s processed: 15473040
Info:             Valid C1s: 15262630
Info:           Invalid C1s: 210410
Info:         C1s corrected: 3165908
Info:  Delay buffer flushes: 1
Info:         C1 Error rate: 21.8207%
Info: 
Info: F3 to F2 frame C2 Error correction:
Info:   Total C2s processed: 13111283
Info:             Valid C2s: 13065477
Info:           Invalid C2s: 45806
Info:         C2s corrected: 804367
Info:  Delay buffer flushes: 1
Info: 
Info: F3 to F2 frame C2 Deinterleave:
Info:   Total C2s processed: 12306912
Info:             Valid C2s: 12221910
Info:           Invalid C2s: 85002
Info:  Delay buffer flushes: 1
Info: 
Info: F2 Frames to F1 Frames:
Info:             Valid frames: 12221848
Info:           Invalid frames: 84992
Info:   Missing section frames: 8179570 ( 1363261 F3 Frames )
Info:       Encoder off frames: 0
Info:             TOTAL frames: 20486410
Info: 
Info:        Frames start time: 00:00.01
Info:          Frames end time: 46:27.20
Info: 
Info: F1 Frames to Audio:
Info:        Audio samples: 73331088
Info:      Corrupt samples: 0
Info:      Missing samples: 49077420
Info:    Concealed samples: 509952
Info:        Total samples: 122918460
Info: 
Info:     Audio start time: 00:00.01
Info:   Audio current time: 46:27.20
Info:       Audio duration: 46:27.19
Info: 
Info: F1 Frames to Data:
Info:          Valid sectors: 0
Info:        Invalid sectors: 0
Info:        Missing sectors: 0
Info:          Total sectors: 0
Info: 
Info:   Sectors missing sync: 0
Info: 
Info:          Start address: 00:00.00
Info:        Current address: 00:00.00

```

Audio start time at 00:00.01, while initial disc time is 18:14.43 leads to 18:14.43 of padded silence before the audio starts.

blindrezo, 2019-09-24:
```
ld-process-efm --debug
Debug: Configuration::readConfiguration(): Reading configuration from "/home/blindrezo/.config/ld-process-efm.ini"
Debug: Configuration::readConfiguration(): sourceDirectory = "/media/blindrezo/8TB/ld-decode"
Debug: Configuration::readConfiguration(): audioDirectory = "/media/blindrezo/8TB/ld-decode"
Debug: Configuration::readConfiguration(): dataDirectory = "/home/blindrezo"
Debug: MainWindow::on_actionOpen_EFM_file_triggered(): Called
Gtk-Message: 00:12:53.572: GtkDialog mapped without a transient parent. This is discouraged.
Debug: MainWindow::on_actionOpen_EFM_file_triggered(): Setting EFM source directory to: "/media/blindrezo/8TB/ld-decode"
Debug: Configuration::writeConfiguration(): sourceDirectory = "/media/blindrezo/8TB/ld-decode"
Debug: Configuration::writeConfiguration(): audioDirectory = "/media/blindrezo/8TB/ld-decode"
Debug: Configuration::writeConfiguration(): dataDirectory = "/home/blindrezo"
Debug: Configuration::writeConfiguration(): Writing configuration to disk
Debug: MainWindow::on_actionOpen_TBC_file_triggered(): Set current file name to to: "/media/blindrezo/8TB/ld-decode/Fam1Side2.efm"
Debug: MainWindow::on_decodePushButton_clicked(): Called
Debug: MainWindow::on_decodePushButton_clicked(): Opened EFM input file
Debug: MainWindow::on_decodePushButton_clicked(): Opened audio output temporary file
Debug: MainWindow::on_decodePushButton_clicked(): Opened data output temporary file
Debug: EfmProcess::setAudioErrorTreatment(): Audio error treatment set to conceal
Debug: EfmProcess::setAudioErrorTreatment(): Conceal type set to linear interpolation
Debug: EfmProcess::setDecoderOptions(): Pad initial disc time is false
Debug: EfmProcess::setDecoderOptions(): Decode as audio is true
Debug: EfmProcess::setDecoderOptions(): Decode as data is false
Debug: EfmProcess::run(): Thread initial run
Debug: EfmProcess::run(): Thread wake up on restart
Debug: EfmProcess::run(): Thread sleeping pending restart
Debug: MainWindow::processingCompleteSignalHandler(): Processing complete - audio available
Info: 
Info: EFM to F3 Frames:
Info:             Valid syncs: 12470319
Info:         Overshoot syncs: 3154
Info:        Undershoot syncs: 5907
Info:             TOTAL syncs: 12479380
Info: 
Info:       Valid EFM symbols: 411512362
Info:     Invalid EFM symbols: 71029
Info:          EFM error rate: 0.0172575%
Info: 
Info:       In range T-values: 1518643709
Info:   Out of range T-values: 0
Info:          TOTAL T-values: 1518643709
Info: 
Info:            Valid frames: 12465579
Info:        Overshoot frames: 6939
Info:       Undershoot frames: 6863
Info:            TOTAL frames: 12479381
Info: 
Info: F3 Frame synchronisation:
Info:    Total input F3 Frames: 12479381
Info:         Discarded Frames: 3107
Info:     Total valid sections: 127308 ( 12476184 F3 Frames )
Info: 
Info: F3 Frame to F2 Frame decode:
Info:       Total input F3 Frames: 12476184
Info:      Total output F2 Frames: 12474224
Info:         Total Preemp Frames: 0
Info:   F3 Sequence Interruptions: 16
Info:           Missing F3 Frames: 29367954
Info:           Initial disc time: 24:52.12
Info:             Final disc time: 28:16.54
Info: 
Info: F3 to F2 frame C1 Error correction:
Info:   Total C1s processed: 14097088
Info:             Valid C1s: 14040508
Info:           Invalid C1s: 56580
Info:         C1s corrected: 1620921
Info:  Delay buffer flushes: 16
Info:         C1 Error rate: 11.8996%
Info: 
Info: F3 to F2 frame C2 Error correction:
Info:   Total C2s processed: 12761964
Info:             Valid C2s: 12749912
Info:           Invalid C2s: 12052
Info:         C2s corrected: 287633
Info:  Delay buffer flushes: 16
Info: 
Info: F3 to F2 frame C2 Deinterleave:
Info:   Total C2s processed: 12474297
Info:             Valid C2s: 12452971
Info:           Invalid C2s: 21326
Info:  Delay buffer flushes: 16
Info: 
Info: F2 Frames to F1 Frames:
Info:             Valid frames: 12452898
Info:           Invalid frames: 21326
Info:   Missing section frames: 37137688 ( 6189614 F3 Frames )
Info:       Encoder off frames: 17542
Info:             TOTAL frames: 49611912
Info: 
Info:        Frames start time: 24:52.12
Info:          Frames end time: 28:16.53
Info: 
Info: F1 Frames to Audio:
Info:        Audio samples: 74717388
Info:      Corrupt samples: 0
Info:      Missing samples: 157022460
Info:    Concealed samples: 127956
Info:        Total samples: 231867804
Info: 
Info:     Audio start time: 24:52.12
Info:   Audio current time: 28:16.53
Info:       Audio duration: 03:24.41
Info: 
Info: F1 Frames to Data:
Info:          Valid sectors: 0
Info:        Invalid sectors: 0
Info:        Missing sectors: 0
Info:          Total sectors: 0
Info: 
Info:   Sectors missing sync: 0
Info: 
Info:          Start address: 00:00.00
Info:        Current address: 00:00.00
```
Audio start time at 24:52.12, but the audio only starts at 59:24. Also, duration shows 3:24.41 while it should really be 28:17, but it's roughly 1 hour and a half.

IanSmallshire, 2019-11-02:
[EMF.log](https://github.com/happycube/ld-decode/files/3800328/EMF.log)
Audio start time is reported as 57.47.52 and its padding the output file by that amount.
During processingthe audio durection counts from a negative number.
Output looks good apart from the padding and plays well for a CLV PAL disc.

simoninns, 2019-12-22:
Fixed in latest commit.

closed 2019-12-22 by simoninns