happycube/ld-decode #343: Vroop - Hsync lost all throughout the disc except first 3k frames

IanSmallshire, 2019-11-01:
Same issue as #252 but further on in the same disc.

There is almost a H sync loss every few seconds throughout the entire disc after frame 3k
[image]

This is one of the Minor Vroops
[image]

Have supplied a 100 frame sample in various locations across the disc. 
https://drive.google.com/drive/folders/1C5ajrO9LmxaDO5y-VSOs2VRznP2BcVrq

simoninns, 2019-12-22:
@IanSmallshire  - was this and #252 resolved?  Can I close both issues?

simoninns, 2019-12-22:
Wrong button :)

IanSmallshire, 2019-12-22:
It is still an issue but have only found one disc affected. 
I have the other discs from BBC so will cap and retest other sides too.

IanSmallshire, 2020-04-15:
Seems fixed after #218 fix

closed 2020-04-17 by happycube