happycube/ld-decode #347: Luma mode blurry compared to source, chroma

Gamnn, 2019-11-09:
The filter for luma seems to need some adjustment, it's blurry compared to the ntsc comb output.

Frame 18746 from ggv-1069:

[image]
[image]
[image]

simoninns, 2019-11-09:
As mentioned on IRC, the luma mode in ld-analyse isn't using the comb filter and is there to help with development (of things like the diffdod tool).  It might be possible to make the filter a little sharper, but there is no need for it since the ld-chroma-decoder has a really good B&W mode for actual video output

atsampson, 2019-11-10:
It might be better to call that mode "LPF" rather than "Luma"? It's never going to look particularly sharp, even if the cutoff is made sharper, because the idea is to remove the HF content (which includes both noise and legitimate sharp edges).

simoninns, 2019-11-11:
Re-opening issue - Agreed that the Luma mode should be renamed to save confusion.  I'll fix it and then close this issue.

simoninns, 2019-11-11:
Fixed in latest commit.

closed 2019-11-11 by simoninns