happycube/ld-decode #350: Rev 6 build error

toynbeeidea, 2019-11-19:
`In file included from mainwindow.h:40:0,
                 from main.cpp:25:
dropoutanalysisdialog.h:29:10: fatal error: qwt/qwt_plot.h: No such file or directory
 #include <qwt/qwt_plot.h>
          ^~~~~~~~~~~~~~~~
compilation terminated.
Makefile:748: recipe for target 'main.o' failed
make[1]: *** [main.o] Error 1
make[1]: Leaving directory '/home/erik/ld-decode/tools/ld-analyse'
Makefile:59: recipe for target 'sub-ld-analyse-all' failed
make: *** [sub-ld-analyse-all] Error 2
`

closed 2019-11-19 by toynbeeidea

toynbeeidea, 2019-11-19:
nvm, I dunno if a new requiremet got added when I wasn't looking but I was missing one.  Derp.

simoninns, 2019-11-19:
This was a new requirement issued recently.  Since rev6 is a dev branch only the main installation documentation is updated (although new requirements are noted in the commit messages for other devs)