happycube/ld-decode #351: #325, #234 still not quite fixed in pre-release 6

toynbeeidea, 2019-11-19:
Closing comment on #325 was that I could reopen, but I don't have access rights to do that, so here's a new filing entirely.

Tried out ld-decode v6 on Moomin (1969) vol 5, and its still a mess.  Here is a five-second sample:

https://drive.google.com/open?id=1437iZsw3biyLEieHqbNZRsKzPOHc1un4

happycube, 2019-12-05:
Finally took a look at this, it's caused by the phase offset being off spec.  The workaround code that does a 90 degree phase adjustment worked fine for #234, but not for #325 and this sample.

I've changed it so the workaround code will only trigger if the disk is *seriously* off, and when it does trigger it does the correct action for 325 and this sample.

toynbeeidea, 2019-12-08:
Hello!  I have rebuilt v6, and yes, it seems okay so far.  I haven't done the whole disc yet, I'll try it overnight and get back to you tomorrow.

Gamnn, 2019-12-10:
Video appears good in my testing, but EFM appears to break in a previously affected area.

[okotowari.ldf](https://drive.google.com/open?id=1imZY5Jzq9xXgN-eJ0FCq615QtO3QVQg8)

toynbeeidea, 2019-12-11:
Yeah, I didn't even check the EFM, never really got it to work right and am not really worried about it as much at this stage.  I didn't notice any video issues in a full decode of vol 5 of Moomin, but if Gamnn feels this is still an issue with regards to the EFM, I won't close this one quite yet.

toynbeeidea, 2020-01-03:
Closing as covered in #386

closed 2020-01-03 by toynbeeidea