happycube/ld-decode #352: EFM regression

Gamnn, 2019-11-25:
EFM seems to have regressed.

4373186:
```
Info: EFM to F3 Frames:
Info:             Valid syncs: 13236128
Info:         Overshoot syncs: 2321
Info:        Undershoot syncs: 5024
Info:             TOTAL syncs: 13243473
Info: 
Info:       Valid EFM symbols: 436739366
Info:     Invalid EFM symbols: 31320
Info:          EFM error rate: 0.00717081%
Info: 
Info:       In range T-values: 1596587924
Info:   Out of range T-values: 0
Info:          TOTAL T-values: 1596587924
Info: 
Info:            Valid frames: 13231517
Info:        Overshoot frames: 5995
Info:       Undershoot frames: 5962
Info:            TOTAL frames: 13243474
Info: 
Info: F3 Frame synchronisation:
Info:    Total input F3 Frames: 13243474
Info:         Discarded Frames: 624
Info:     Total valid sections: 135077 ( 13237546 F3 Frames )
Info: 
Info: F3 Frame to F2 Frame decode:
Info:       Total input F3 Frames: 13237546
Info:      Total output F2 Frames: 13237252
Info:         Total Preemp Frames: 0
Info:   F3 Sequence Interruptions: 1
Info:           Missing F3 Frames: 5880
Info:           Initial disc time: 00:00.00
Info:             Final disc time: 30:01.61
Info: 
Info: F3 to F2 frame C1 Error correction:
Info:   Total C1s processed: 13671678
Info:             Valid C1s: 13658869
Info:           Invalid C1s: 12809
Info:         C1s corrected: 434134
Info:  Delay buffer flushes: 1
Info:         C1 Error rate: 3.26912%
Info: 
Info: F3 to F2 frame C2 Error correction:
Info:   Total C2s processed: 13386873
Info:             Valid C2s: 13386870
Info:           Invalid C2s: 3
Info:         C2s corrected: 149545
Info:  Delay buffer flushes: 1
Info: 
Info: F3 to F2 frame C2 Deinterleave:
Info:   Total C2s processed: 13237324
Info:             Valid C2s: 13237318
Info:           Invalid C2s: 6
Info:  Delay buffer flushes: 1
Info: 
Info: F2 Frames to F1 Frames:
Info:             Valid frames: 13237246
Info:           Invalid frames: 6
Info:   Missing section frames: 6076 ( 1012 F3 Frames )
Info:       Encoder off frames: 14602
Info:             TOTAL frames: 13243328
Info: 
Info:        Frames start time: 00:00.00
Info:          Frames end time: 30:01.60
Info: 
Info: F1 Frames to Audio:
Info:        Audio samples: 79423476
Info:      Corrupt samples: 0
Info:      Missing samples: 36456
Info:    Concealed samples: 36
Info:        Total samples: 79459968
Info: 
Info:     Audio start time: 00:00.00
Info:   Audio current time: 30:01.60
Info:       Audio duration: 30:01.60
Info: 
Info: F1 Frames to Data:
Info:          Valid sectors: 0
Info:        Invalid sectors: 0
Info:        Missing sectors: 0
Info:          Total sectors: 0
Info: 
Info:   Sectors missing sync: 0
Info: 
Info:          Start address: 00:00.00
Info:        Current address: 00:00.00
```

191d638:
```
Info: EFM to F3 Frames:
Info:             Valid syncs: 13190714
Info:         Overshoot syncs: 47729
Info:        Undershoot syncs: 5029
Info:             TOTAL syncs: 13243472
Info: 
Info:       Valid EFM symbols: 436554839
Info:     Invalid EFM symbols: 218253
Info:          EFM error rate: 0.0499694%
Info: 
Info:       In range T-values: 1596585740
Info:   Out of range T-values: 0
Info:          TOTAL T-values: 1596585740
Info: 
Info:            Valid frames: 13186214
Info:        Overshoot frames: 29094
Info:       Undershoot frames: 28165
Info:            TOTAL frames: 13243473
Info: 
Info: F3 Frame synchronisation:
Info:    Total input F3 Frames: 13243473
Info:         Discarded Frames: 84413
Info:     Total valid sections: 133737 ( 13106226 F3 Frames )
Info: 
Info: F3 Frame to F2 Frame decode:
Info:       Total input F3 Frames: 13106226
Info:      Total output F2 Frames: 13102306
Info:         Total Preemp Frames: 0
Info:   F3 Sequence Interruptions: 34
Info:           Missing F3 Frames: 137200
Info:           Initial disc time: 00:00.00
Info:             Final disc time: 30:01.61
Info: 
Info: F3 to F2 frame C1 Error correction:
Info:   Total C1s processed: 13537012
Info:             Valid C1s: 13515292
Info:           Invalid C1s: 21720
Info:         C1s corrected: 430821
Info:  Delay buffer flushes: 34
Info:         C1 Error rate: 3.34299%
Info: 
Info: F3 to F2 frame C2 Error correction:
Info:   Total C2s processed: 13251670
Info:             Valid C2s: 13239046
Info:           Invalid C2s: 12624
Info:         C2s corrected: 149259
Info:  Delay buffer flushes: 34
Info: 
Info: F3 to F2 frame C2 Deinterleave:
Info:   Total C2s processed: 13102341
Info:             Valid C2s: 13089627
Info:           Invalid C2s: 12714
Info:  Delay buffer flushes: 34
Info: 
Info: F2 Frames to F1 Frames:
Info:             Valid frames: 13089592
Info:           Invalid frames: 12714
Info:   Missing section frames: 141022 ( 23503 F3 Frames )
Info:       Encoder off frames: 14602
Info:             TOTAL frames: 13243328
Info: 
Info:        Frames start time: 00:00.00
Info:          Frames end time: 30:01.60
Info: 
Info: F1 Frames to Audio:
Info:        Audio samples: 78537552
Info:      Corrupt samples: 0
Info:      Missing samples: 846132
Info:    Concealed samples: 76284
Info:        Total samples: 79459968
Info: 
Info:     Audio start time: 00:00.00
Info:   Audio current time: 30:01.60
Info:       Audio duration: 30:01.60
Info: 
Info: F1 Frames to Data:
Info:          Valid sectors: 0
Info:        Invalid sectors: 0
Info:        Missing sectors: 0
Info:          Total sectors: 0
Info: 
Info:   Sectors missing sync: 0
Info: 
Info:          Start address: 00:00.00
Info:        Current address: 00:00.00
```

happycube, 2019-11-26:
Can I look at a ~10 second sample please?  ralf doesn't reproduce it (but I'll try other snippets)

happycube, 2019-11-26:
Got the sample off-github, able to reproduce.  Will need to upgrade some line # locating code. :)

happycube, 2019-12-01:
Should be fixed now

closed 2019-12-01 by happycube