happycube/ld-decode #366: Record per-field RF sample numbers in JSON

atsampson, 2019-12-05:
Given a field from a .tbc, there's currently no direct way to locate the corresponding samples in the RF input. We could fix this by recording the input sample number corresponding to the start of each field in the JSON.

I can think of a few potential uses for this:

- Filling gaps in EFM samples. Some discs don't have CD-style timestamps in the EFM, so this would let an EFM-combining tool (#198) lock to the video instead.

- Rerunning EFM or analogue audio decodes without needing to decode the video again.

- Showing the RF waveform in ld-analyse's scope ("what did that dropout actually look like?").

- Aligning separately-captured audio (e.g. for VHS) with decoded video.

It looks like the implementation would be a one-liner: `LDdecode.buildmetadata` already computes a low-res `diskLoc` value in `-s` units, so it could just store `self.fieldLoc / 2` as another value.

happycube, 2019-12-24:
I kinda missed this one before finalizing rev6, but it was enabled at one point but Simon wanted the JSON simplified for processing/size reasons.

However, I think this makes it a 2-1 vote, so it can go back in ;)

simoninns, 2019-12-24:
Add it back in.  No problem for me.  JSON metadata performance in the tools seems to be under control at the moment.

simoninns, 2019-12-24:
Don't forget to update the JSON metadata wiki though ;)

happycube, 2019-12-24:
Done, and added an option to seek back to the same spot (although it won't decode from *exactly* the same locations)

commit 11fc1687 (2019-12-24): fix #366, and add --start_fileloc option to (mostly) restart at the same place
closed 2019-12-24 by happycube