happycube/ld-decode #371: ld-process-efm decoding failure on some datasets...

happycube, 2019-12-18:
I have a bit from a decode of the Twin Peaks Visual Soundtrack (no rot, solid SNR) that completely fails at timestamp 6:14.14 - a full disc decode completely cuts out at this point.  

Short (4 second) .efm sample is at https://drive.google.com/open?id=1SFXXFAq6YogOZV4TNekJVpKnAlSoeOmt - it shows the failure nicely, and chopping the first 1MB off the file allows the end of it to decode cleanly, showing that the .efm file has a momentary glitch.

atsampson, 2019-12-18:
This seems to be `EfmToF3Frames::sm_state_findSecondSync` getting confused by a mangled frame. It decodes fine until it hits a "F3 Sync value low and off by one transition backwards" problem, then its idea of where the syncs are never gets reset.

I tried adding `return state_findInitialSyncStage1;` at the end of the "// Handle various possible sync issues in a (hopefully) smart way" block, so it resyncs from scratch when it sees a dodgy end sync - with that change, it gets back on track pretty quickly and only drops 2352 samples. @simoninns might have a better fix?

simoninns, 2019-12-21:
I've added some additional logic to track the number of good syncs between bad syncs.  If the sync is failing and no good syncs preceded, then the code switches state back to finding the initial sync.  This seems to cause a much faster reset and stops the code from getting into a loop where it never quite recovers sync.  Needs a bit of regression testing.

atsampson, 2019-12-21:
Sound good to me.

simoninns, 2019-12-22:
Tested with the twinpeaks full disc on the latest build and it decoded fine.  There are some sync-loss gaps, but that seems to be upstream (errors in the disc image).

closed 2019-12-22 by simoninns