happycube/ld-decode #381: Two screens of Python babble on ctrl-c

simoninns, 2019-12-23:
Terminating a running ld-decode process results in the following output:

`file frame 14502 CAV frame 14203
^CProcess Process-2:
Process Process-1:
Traceback (most recent call last):
  File "/usr/lib/python3.6/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python3.6/multiprocessing/process.py", line 93, in run
    self._target(*self._args, **self._kwargs)
  File "/home/sdi/github/ld-decode/lddecode_core.py", line 788, in worker
    output['demod'] = self.rf.demodblock(fftdata = fftdata, mtf_level=target_MTF, cut=True)
Traceback (most recent call last):
  File "/home/sdi/github/ld-decode/lddecode_core.py", line 472, in demodblock
    video_out = np.rec.array([out_video, demod, demod_hpf, out_video05, out_videopilot], names=['demod', 'demod_raw', 'demod_hpf', 'demod_05', 'demod_pilot'])
  File "/usr/lib/python3/dist-packages/numpy/core/records.py", line 826, in array
    return fromarrays(obj, dtype=dtype, shape=shape, **kwds)
  File "/usr/lib/python3/dist-packages/numpy/core/records.py", line 625, in fromarrays
    _array[_names[i]] = arrayList[i]
  File "/usr/lib/python3.6/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python3.6/multiprocessing/process.py", line 93, in run
    self._target(*self._args, **self._kwargs)
  File "/home/sdi/github/ld-decode/lddecode_core.py", line 788, in worker
    output['demod'] = self.rf.demodblock(fftdata = fftdata, mtf_level=target_MTF, cut=True)
  File "/home/sdi/github/ld-decode/lddecode_core.py", line 454, in demodblock
    hilbert = npfft.ifft(indata_fft_filt)
KeyboardInterrupt
  File "/usr/lib/python3/dist-packages/numpy/fft/fftpack.py", line 287, in ifft
    output = _raw_fft(a, n, axis, fftpack.cffti, fftpack.cfftb, _fft_cache)
  File "/usr/lib/python3/dist-packages/numpy/fft/fftpack.py", line 83, in _raw_fft
    r = work_function(a, wsave)
KeyboardInterrupt
Process Process-3:
Process Process-4:
Traceback (most recent call last):
  File "/usr/lib/python3.6/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python3.6/multiprocessing/process.py", line 93, in run
    self._target(*self._args, **self._kwargs)
  File "/home/sdi/github/ld-decode/lddecode_core.py", line 788, in worker
    output['demod'] = self.rf.demodblock(fftdata = fftdata, mtf_level=target_MTF, cut=True)
  File "/home/sdi/github/ld-decode/lddecode_core.py", line 452, in demodblock
    indata_fft_filt *= self.Filters['MTF'] ** mtf_level
KeyboardInterrupt
Traceback (most recent call last):
  File "/usr/lib/python3.6/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python3.6/multiprocessing/process.py", line 93, in run
    self._target(*self._args, **self._kwargs)
  File "/home/sdi/github/ld-decode/lddecode_core.py", line 767, in worker
    if pipein.poll():
  File "/usr/lib/python3.6/multiprocessing/connection.py", line 257, in poll
    return self._poll(timeout)
  File "/usr/lib/python3.6/multiprocessing/connection.py", line 414, in _poll
    r = wait([self], timeout)
  File "/usr/lib/python3.6/multiprocessing/connection.py", line 911, in wait
    ready = selector.select(timeout)
  File "/usr/lib/python3.6/selectors.py", line 376, in select
    fd_event_list = self._poll.poll(timeout)
KeyboardInterrupt
Terminated, saving JSON and exiting
^CError in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python3.6/multiprocessing/util.py", line 262, in _run_finalizers
    finalizer()
  File "/usr/lib/python3.6/multiprocessing/util.py", line 186, in __call__
    res = self._callback(*self._args, **self._kwargs)
  File "/usr/lib/python3.6/multiprocessing/queues.py", line 191, in _finalize_join
    thread.join()
  File "/usr/lib/python3.6/threading.py", line 1056, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.6/threading.py", line 1072, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt
`

A simple "process terminated by user" message would probably be nicer :)

happycube, 2019-12-23:
Yup, It Does That.  (I need to figure out how to make it Not Do That.)

happycube, 2019-12-24:
A bit of google-fu found https://stackoverflow.com/questions/11312525/catch-ctrlc-sigint-and-exit-multiprocesses-gracefully-in-python which I applied to ld-decode.py and ld-cut.py, seems to work.

commit 0c6aa5d1 (2019-12-24): Fix #381
closed 2019-12-24 by happycube