happycube/ld-decode #386: EFM breaks on some discs

Gamnn, 2019-12-26:
EFM breaks on this set of LDs, with around 20% C1 error rate. The discs were formerly affected by hedgehogging ala #351, #325 et al. which may or may not be related.

sample: [okotowari.ldf](https://drive.google.com/open?id=1imZY5Jzq9xXgN-eJ0FCq615QtO3QVQg8)

Gamnn, 2019-12-26:
Capped with a different player. and the section decodes fine. [Okotowari-919.ldf](https://drive.google.com/open?id=1CH1wZqFJMghK6ZAUJzfoXLVwTbqpu6CV)

However, the digital audio played on the first player, so I would assume it could still be decoded with some tuning.

simoninns, 2019-12-26:
Were both players standard LD players?

Gamnn, 2019-12-26:
first was HLD-X9, second DVL-919.

simoninns, 2019-12-31:
Do we think this is a real issue, or more to do with the X9 being kinda broken?

Gamnn, 2019-12-31:
I'd say it's a mix. The x9 decodes the audio in this section (mostly) fine, so ld-decode and tools should be able to as well.

atsampson, 2019-12-31:
The filtered EFM waveform looks pretty horrible in both of those samples. (As do the VBI/VITS waveforms, but that's probably a different problem.)

919:
[image]

X9:
[image]

Adjusting the filter shape doesn't seem to improve the X9 sample much...

simoninns, 2020-01-01:
Unless these test samples are coming from players that have been verified as calibrated (using GGV1069) I don't think it's really worth digging into as there are too many environmental variables to really understand what's going on.  If the EFM signal is bad, it's bad.  Marking as wontfix for now.

simoninns, 2020-01-24:
Closing issue as wontfix

closed 2020-01-24 by simoninns