happycube/ld-decode #388: Loss of Sync mid disc causing 1 second skip in RF

IanSmallshire, 2019-12-29:
From the middle of a decode:

file frame 19399 CLV timecode 10:36.13 frame 19093
file frame 19400 CLV timecode 10:36.14 frame 19094
file frame 19401 CLV timecode 10:36.15 frame 19095
file frame 19402 CLV timecode 10:36.16 frame 19096
file frame 19403 CLV timecode 10:36.17 frame 19097
file frame 19404 CLV timecode 10:36.18 frame 19098
file frame 19405 CLV timecode 10:36.19 frame 19099
ERROR:root:Unable to find any sync pulses, jumping one second
ERROR:root:Unable to find any sync pulses, jumping one second
file frame 19466 CLV timecode 10:38.20 frame 19160
file frame 19467 CLV timecode 10:38.21 frame 19161
file frame 19468 CLV timecode 10:38.22 frame 19162
file frame 19469 CLV timecode 10:38.23 frame 19163
file frame 19470 CLV timecode 10:38.24 frame 19164
file frame 19471 CLV timecode 10:38.25 frame 19165

From Happycube on IRC:
looks like the carrier's dipped low, this disk needs AGC - Because Technidisc(tm)

sample supplied of 100 frames

IanSmallshire, 2019-12-29:
Various errors distributed across the disc that are probably all related due to incorrect levels in mastering.

Happy to supply samples if needed

It usually skips 2 seconds before locking on again but sometimes has a single frame whet the lock is good between the two second jumps.

file frame 4133 CLV timecode 2:07.02 frame 3812
file frame 4134 CLV timecode 2:07.03 frame 3813
/home/ian/.local/lib/python3.6/site-packages/numpy/core/fromnumeric.py:3118: RuntimeWarning: Mean of empty slice.
  out=out, **kwargs)
/home/ian/.local/lib/python3.6/site-packages/numpy/core/_methods.py:85: RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)
ERROR:root:Unable to find any sync pulses, jumping one second
file frame 4165 CLV timecode 2:08.04 frame 3844
file frame 4166 CLV timecode 2:08.05 frame 3845
...
file frame 4685 CLV timecode 2:25.14 frame 4364
file frame 4686 CLV timecode 2:25.15 frame 4365
ERROR:root:Unable to find any sync pulses, jumping one second
file frame 4717 CLV timecode 2:26.16 frame 4396
file frame 4718 CLV timecode 2:26.17 frame 4397
ERROR:root:Unable to find any sync pulses, jumping one second
file frame 4749 CLV timecode 2:27.18 frame 4428
file frame 4750 CLV timecode 2:27.19 frame 4429
...
file frame 8387 CLV timecode 4:29.00 frame 8070
file frame 8388 CLV timecode 4:29.01 frame 8071
ERROR:root:Unable to find any sync pulses, jumping one second
file frame 8419 CLV timecode 4:29.02 frame 8072
file frame 8420 CLV timecode 4:30.04 frame 8104
...
file frame 15915 CLV timecode 8:40.06 frame 15606
file frame 15916 CLV timecode 8:40.07 frame 15607
file frame 15917 CLV timecode 8:40.08 frame 15608
ERROR:root:Unable to find any sync pulses, jumping one second
WARNING:root:Severe VSYNC-area corruption detected.
ERROR:root:Extreme VSYNC-area corruption detected, dropping field
ERROR:root:Unable to determine start of field - dropping field
WARNING:root:Severe VSYNC-area corruption detected.
ERROR:root:Extreme VSYNC-area corruption detected, dropping field
ERROR:root:Unable to determine start of field - dropping field
WARNING:root:Severe VSYNC-area corruption detected.
ERROR:root:Extreme VSYNC-area corruption detected, dropping field
ERROR:root:Unable to determine start of field - dropping field
WARNING:root:Severe VSYNC-area corruption detected.
ERROR:root:Extreme VSYNC-area corruption detected, dropping field
ERROR:root:Unable to determine start of field - dropping field
WARNING:root:Severe VSYNC-area corruption detected.
ERROR:root:Extreme VSYNC-area corruption detected, dropping field
ERROR:root:Unable to determine start of field - dropping field
WARNING:root:Severe VSYNC-area corruption detected.
ERROR:root:Extreme VSYNC-area corruption detected, dropping field
ERROR:root:Unable to determine start of field - dropping field
WARNING:root:Severe VSYNC-area corruption detected.
ERROR:root:Extreme VSYNC-area corruption detected, dropping field
ERROR:root:Unable to determine start of field - dropping field
WARNING:root:Severe VSYNC-area corruption detected.
ERROR:root:Extreme VSYNC-area corruption detected, dropping field
ERROR:root:Unable to determine start of field - dropping field
WARNING:root:Severe VSYNC-area corruption detected.
ERROR:root:Extreme VSYNC-area corruption detected, dropping field
ERROR:root:Unable to determine start of field - dropping field
WARNING:root:Severe VSYNC-area corruption detected.
ERROR:root:Extreme VSYNC-area corruption detected, dropping field
ERROR:root:Unable to determine start of field - dropping field
WARNING:root:Severe VSYNC-area corruption detected.
ERROR:root:Extreme VSYNC-area corruption detected, dropping field
ERROR:root:Unable to determine start of field - dropping field
WARNING:root:Severe VSYNC-area corruption detected.
ERROR:root:Extreme VSYNC-area corruption detected, dropping field
ERROR:root:Unable to determine start of field - dropping field
WARNING:root:Severe VSYNC-area corruption detected.
ERROR:root:Extreme VSYNC-area corruption detected, dropping field
ERROR:root:Unable to determine start of field - dropping field
WARNING:root:Severe VSYNC-area corruption detected.
ERROR:root:Extreme VSYNC-area corruption detected, dropping field
ERROR:root:Unable to determine start of field - dropping field
ERROR:root:Unable to find any sync pulses, jumping one second
WARNING:root:NTSC field phaseID sequence mismatch (player may be paused)
ERROR:root:Skipped field
file frame 15984 CLV timecode 8:40.09 frame 15609
file frame 15985 CLV timecode 8:42.16 frame 15676
file frame 15986 CLV timecode 8:42.17 frame 15677
file frame 15987 CLV timecode 8:42.18 frame 15678
...
file frame 18198 CLV timecode 9:56.11 frame 17891
ERROR:root:Unable to find any sync pulses, jumping one second
file frame 18229 CLV timecode 9:56.12 frame 17892
...
file frame 20783 CLV timecode 11:22.18 frame 20478
file frame 20784 CLV timecode 11:22.19 frame 20479
ERROR:root:Unable to find any sync pulses, jumping one second
file frame 20815 CLV timecode 11:23.20 frame 20510
ERROR:root:Unable to find any sync pulses, jumping one second
file frame 20846 CLV timecode 11:24.21 frame 20541
file frame 20847 CLV timecode 11:24.22 frame 20542
...
file frame 29340 CLV timecode 16:08.03 frame 29043
file frame 29341 CLV timecode 16:08.04 frame 29044
ERROR:root:Unable to find any sync pulses, jumping one second
ERROR:root:Unable to find any sync pulses, jumping one second
file frame 29402 CLV timecode 16:10.05 frame 29105
file frame 29403 CLV timecode 16:10.06 frame 29106
...
file frame 30262 CLV timecode 16:38.26 frame 29966
file frame 30263 CLV timecode 16:38.27 frame 29967
ERROR:root:Unable to find any sync pulses, jumping one second
ERROR:root:Unable to find any sync pulses, jumping one second
file frame 30323 CLV timecode 16:38.28 frame 29968
file frame 30325 CLV timecode 16:40.29 frame 30029

simoninns, 2020-01-01:
It might be better @happycube to use a 1 second skip when starting a capture; but reduce this to a much smaller skip in the event of mid-decode sync loss?

happycube, 2020-01-01:
That sounds good to me... I'll probably change it to a 1-2 field skip once
it's decoded a few frames.

Gamnn, 2020-04-30:
Seems related to #264

commit 308f59cf (2022-02-06): Merge pull request #701 from happycube/chad-2022.02.06

fix #388: only skip one field after something's been decoded already
closed 2022-02-06 by happycube