happycube/ld-decode #389: downscale_audio crash on TBC failure

atsampson, 2019-12-30:
While decoding [EE 1015](https://www.lddb.com/laserdisc/19061/EE-1015/Hard-Boiled), using rev 856f2c0ee952b468e11cbe8ae758eaabb865be35:

```
file frame 61066 CLV timecode 40:35.15 frame 60890
file frame 61067 CLV timecode 40:35.16 frame 60891
file frame 61068 CLV timecode 40:35.17 frame 60892
file frame 61069 CLV timecode 40:35.18 frame 60893
WARNING:root:WARNING: TBC failure at line 313
ERROR - please paste the following into a bug report:
current sample: 97712956725.25229
arguments: Namespace(MTF=None, MTF_offset=None, daa=False, ignoreleadout=False, infile='HardBoiled_CLV_PAL_side1_4400_2019-10-11_20-22-17.raw.oga', inputfreq=None, length=110000, nodod=False, noefm=False, ntsc=False, ntscj=False, outfile='out/HardBoiled_CLV_PAL_side1_4400_2019-10-11_20-22-17', pal=True, seek=-1, start=598.0, start_fileloc=-1, threads=5, vbpf_high=None, verboseVITS=False, vlpf=None)
Exception: division by zero  Traceback:
  File "/home/ats/pkg/ld-decode/bin/ld-decode", line 118, in <module>
    f = ldd.readfield()
  File "/home/ats/src/ld-decode/lddecode/core.py", line 2499, in readfield
    picture, audio, efm = f.downscale(linesout = self.output_lines, final=True, audio=self.analog_audio)
  File "/home/ats/src/ld-decode/lddecode/core.py", line 1927, in downscale
    dsout, dsaudio, dsefm = super(FieldPAL, self).downscale(*args, **kwargs)
  File "/home/ats/src/ld-decode/lddecode/core.py", line 1624, in downscale
    self.dsaudio, self.audio_next_offset = downscale_audio(self.data['audio'], lineinfo, self.rf, self.linecount, self.audio_offset, freq = audio)
  File "/home/ats/src/ld-decode/lddecode/core.py", line 990, in downscale_audio
    output_left = nb_mean(audio['audio_left'][np.int(locs[i]):np.int(locs[i+1])])
```

[Here's a 10-frame clip](https://stuff.offog.org/ld/ee1015-tbcfail.raw.oga) around 40:35.19 -- but annoyingly it doesn't reproduce with just that clip...

I've seen other crashes in the past where nb_mean/nb_median get applied to an empty array -- would it make sense to have them check for that and return 0 for robustness (even though it's not the "right" thing to do)?

happycube, 2019-12-30:
Argh, it'd be nice to reproduce the TBC error properly so it doesn't error out.  How far back do you need to go to hit it again?

I created an issue389 branch which will fail safely in this case, leaving null samples.

Also EE1015 does not have analog audio anyhow, so you can just rerun ld-decode with --daa and this code shouldn't even run in the first place...

atsampson, 2019-12-31:
Having rerun a full decode using the same revision with a load of instrumentation added, it doesn't reproduce the TBC failure there either. In particular, it doesn't land on the same fdoffset reported above (and I can't persuade it to do so even by tweaking the fdoffset directly as it approaches that point) - so I'm guessing this was a transient memory or IO error.

Your issue389 branch looks good to me, so OK to close this once that's merged - thanks!

closed 2020-01-07 by happycube