happycube/ld-decode #405: Crash in refine_linelocs_hsync when hsync_area is empty

atsampson, 2020-01-11:
While decoding a clip from the middle of [EE1053](https://www.lddb.com/laserdisc/19051/EE-1053/Nikita):

```
$ ld-decode --pal ee1053-lineloc-crash.raw.oga out/tmp
file frame 1 CLV timecode 41:06.09 frame 61659
file frame 2 CLV timecode 41:06.10 frame 61660
file frame 3 CLV timecode 41:06.11 frame 61661
[...]
file frame 147 CLV timecode 41:12.06 frame 61806
file frame 148 CLV timecode 41:12.07 frame 61807
file frame 149 CLV timecode 41:12.08 frame 61808
ERROR - please paste the following into a bug report:
current sample: 240714034.2967033
arguments: Namespace(MTF=None, MTF_offset=None, WibbleRemover=False, daa=False, ignoreleadout=False, infile='ee1053-lineloc-crash.raw.oga', inputfreq=None, length=110000, lowband=False, nodod=False, noefm=False, ntsc=False, ntscj=False, outfile='out/tmp', pal=True, seek=-1, start=0, start_fileloc=-1, threads=5, vbpf_high=None, verboseVITS=False, vlpf=None)
Exception: zero-size array to reduction operation minimum which has no identity  Traceback:
  File "/home/ats/pkg/ld-decode/bin/ld-decode", line 129, in <module>
    f = ldd.readfield()
  File "/home/ats/src/ld-decode/lddecode/core.py", line 2607, in readfield
    f, offset = self.decodefield(initphase = initphase)
  File "/home/ats/src/ld-decode/lddecode/core.py", line 2590, in decodefield
    f = self.FieldClass(self.rf, self.rawdecode, audio_offset = self.audio_offset, prevfield = self.curfield, initphase = initphase)
  File "/home/ats/src/ld-decode/lddecode/core.py", line 2047, in __init__
    super(FieldPAL, self).__init__(*args, **kwargs)
  File "/home/ats/src/ld-decode/lddecode/core.py", line 1823, in __init__
    self.linelocs2 = self.refine_linelocs_hsync()
  File "/home/ats/src/ld-decode/lddecode/core.py", line 1630, in refine_linelocs_hsync
    if nb_min(hsync_area) < self.rf.iretohz(-55) or nb_max(hsync_area) > self.rf.iretohz(30):
  File "/gar/lib/python3.8/site-packages/numba/targets/arraymath.py", line 502, in array_min_impl
    raise ValueError(MSG)
```

Starting later just produces `ERROR:root:Skipped field` on the offending field.

I've put ee1053-lineloc-crash.raw.oga in sftp:adamsampson/Faults.

happycube, 2020-01-12:
Got it downloaded, investigating slowly.

happycube, 2020-01-12:
Started an issue405 branch.  This is going to take a while as I'm going to clean things up to make them more deterministic (i.e. starting each field decode at a block boundary)

happycube, 2020-01-21:
There's something wrong with the demod here.  That said, it still shouldn't crash. :)

https://drive.google.com/open?id=1VB3UliqrqJlxVCQrFVgxcRhq0fB5zyJm

happycube, 2020-01-21:
This should be fixed now, feel free to reopen if you see otherwise.

closed 2020-01-21 by happycube