happycube/ld-decode #406: Basic usage of ld decode - wiki page is useless

simoninns, 2020-01-12:
Instead of having a few command examples, this wiki page should walk through the whole process of decoding something.  My suggestion would be the Ralf sample; showing each step of the process and describing what it's for.

The walk through should include - ld-decode, ld-process-vbi, ld-dropout-correct, ld-discmap and ld-chroma-decoder piping through ffmpeg to create a final mp4.

Gamnn, 2020-01-15:
A bit of it is covered in the PAL/NTSC decoding guides, but an overall guide on basic workflows applicable to both would be useful. 

A lot of the current examples are outdated, still calling python manually. And a section still refers to the output analog pcm as 48kHz which is no longer the case.

Other parts aren't part of a usual workflow, but give insight into the process and are useful. They should be included or moved into a new page. The 'display' examples for instance show how the .tbc format can be read outside of the ld-decode tools. Likewise the convert examples might be useful in certain cases, but might cause confusion as-written (there's no mention of piping into ffmpeg on the page currently).

toynbeeidea, 2020-01-16:
I wouldn't say it was "useless," though Gamn did have to explain to me that ld-process-vbi and ld-export-metadata allowed me to replicate the chapter layout.  What's there now doesn't explain that sort of thing, at least to a layman like myself.

simoninns, 2020-01-21:
Now the page is awesome again... happy days! :)

closed 2020-01-21 by simoninns