happycube/ld-decode #407: Make ld-process-efm map against VBI

simoninns, 2020-01-13:
On discs such as Domesday AIV the VBI contains a hint as to where data/audio is present (as an EFM track) on the disc.  This could be used to help the tool determine if errors are due to corrupt data or simply missing EFM.

simoninns, 2020-01-13:
An additional thought is to compare time-stamps in the VBI to the EFM sector address - that way you could work around issues where the main TBC has been chopped up a bit (ld-discmap processed for example).

IanSmallshire, 2020-01-13:
In the very Early days of BeebSCSI and I scripted frame captures from a real VP415 I queried the fcode '?D' for every frame and stored and aggregated the results.

National++
start		end		?D
1		45		D;:026
46		808		D;:01?
809		40842	D;:026
40843	54000	D;:01?

simoninns, 2020-01-16:
I tried this, but it didn't really help matters - the vast majority of the missing sections are already dealt with in the EFM processor.  Since the frame rate of the EFM is higher than that of the video (75fps compared to 25/29) the existing missing sector detection already detected everything that the VBI marked.

closed 2020-01-16 by simoninns