happycube/ld-decode #416: ld-analyse jump to chapter

Gamnn, 2020-01-20:
It would be useful if the current start and end frame buttons (⏮ ⏭) in ld-analyse would jump to the next/previous chapter if present in VBI. By holding down, it should skip through them quickly as the next/previous frame buttons do, and include the first and last frame in its range.

simoninns, 2020-01-25:
This is actually a lot more tricky than it seems.  Most discs don't have chapters, some have only 1 chapter, some have chapters in order, some (like National Gallery of Art) have totally random chapter numbers that bounce all over the place (not in numerical order, with repeating numbers).

The only way to solve this usefully is to generate a 'map' which starts at frame 1 and advances through the frames noting whenever the chapter number changes (without bothering to check what it changes to).  These mapped points can then be used to cause the analyser GUI to jump back or forth to the next/previous chapter-change point (which may or may not follow a numerical sequence of chapters)...

Gotta bloody love LDs...

simoninns, 2020-01-25:
Fixed

closed 2020-01-25 by simoninns